ArduPilot/ardupilot_wiki #394: Documentation review in local

imuguruza, 2016-04-22:
I'm not able to see properly the html files generated. I have made next steps:

- `make html`
- `firefox _build/index.html`

But generated index is empty:

[image]

Any idea what I'm doind wrong?

Thanks in advance!

hamishwillee, 2016-04-24:
No idea. What folder are you calling `make html` in? Are you using the Vagrant installation or native?

imuguruza, 2016-04-25:
I'm doinf it in the main folder (`ardupilot_wiki`) in order to build themm all.
How should I use Vagrant in this repo??

hamishwillee, 2016-04-26:
@imuguruza It should work, but I guess you haven't fully set up the environment with the right theme etc. Vagrant is the easiest way to do this. Please follow the instructions here: http://ardupilot.org/dev/docs/common-wiki_editing_guide.html#building-testing-docs-locally

imuguruza, 2016-04-26:
Yeahp you're right, finally I've managed to make it work. On thing I don't like is tahat needs a lot of time for compiling and development is quite slow...

closed 2016-04-26 by imuguruza

hamishwillee, 2016-04-26:
@imuguruza Yes it is a bit slow. This is because we have only one copy of the common files - so they get copied across into the different wikis each time - forcing a full rebuild of those topics. 

You can reduce the time a little bit by only building one site: 
     update.py --side copter

If you're doing multiple builds/tests then you can use update.py for the first build to copy topics and then edit locally and build using `make html` in the different sub folders (e.g. /copter). This is much faster as it is incremental, but you have to remember that any changes to common topics must be copied back to the common folder ... a little risky.

I'd love to make it possible to have the incremental build using the big script, but haven't had time to work out the best way to do that.

imuguruza, 2016-04-27:
Thanks for the suggestion