ArduPilot/ardupilot_wiki #404: RC Transmitter Flight Mode Configuration

Frief84, 2016-04-24:
[RC Transmitter Flight Mode Configuration](http://ardupilot.org/copter/docs/common-rc-transmitter-flight-mode-configuration.html) - Is it possible to make the channel assignment changeable
 like in the microkopter-System ... that would make the Pixhawk easily adaptable!

[image]

magicrub, 2016-04-24:
I'd this a feature request to the firmware? You've posted an issue to the
documentation page...

Frief84, 2016-04-24:
Oh sorry ... yes... i gues it's more a Firmwareissue

especially the changeability of the Flight Mode Chanel would be important!!!
for me ;-)

magicrub, 2016-04-24:
Would you mind closing this issue and opening one in /ardupilot/ardupilot?
Thx.

WickedShell, 2016-04-24:
There are already issues tracking flight channel (and I believe the other
ones as well). The flight mode issue is here:
https://github.com/ArduPilot/ardupilot/issues/3819

davidbuzz, 2016-11-15:
as noted, not a wiki issue.  :-)

closed 2016-11-15 by davidbuzz