ArduPilot/ardupilot_wiki #411: Pixhawk Powering/backup off the servo rail. Voltage should not exceed 5V not 5.7 V

kwikius, 2016-04-25:
The minimum zenner voltage for 1N5339 is 5.3 volts, threfore the input voltage must be less than this , not 5.7 volts as stated in
http://ardupilot.org/copter/docs/common-powering-the-pixhawk.html#powering-backup-off-the-servo-rail
else, due to component tolerance, some zenner may get hot or go pop with a voltage over 5.3 volts and given tolerance of volt meters and psus etc this gives some safety margin

magicrub, 2016-04-25:
@proficnc care to comment?

CraigElder, 2016-04-26:
The input should be supplied with 5V.
5.7v is the maximum input that the voltage selector will accept as a valid input voltage to power the Pixhawk. If there are servos connected to the rail, they can cause the voltage on the rail to fluctuate. If the rail rises above 5.7V the selector will disable the input and turn off the autopilot. the 5.7V zener is intended to clap the voltage and prevent this from occurring.

closed 2016-04-26 by CraigElder

kwikius, 2016-04-26:
Sorry but the warning "At any voltage above 5.7V the servo rail cannot be used to power the Pixhawk and the Zener diode must not be used. The servo rail can supply servos requiring up to 10.5V (but not also power the Pixhawk)." implies that it is OK to power from 5.7 V.. It is not! :)

CraigElder, 2016-04-28:
@kwikius I changed the wording again 

The servo rail can supply servos requiring up to 10.5V (but not also power the Pixhawk). Voltages above 5V cannot be used to power the Pixhawk via the servo rail. In this case the Zener diode **must not be used**. 

https://github.com/ArduPilot/ardupilot_wiki/commit/988ceae42f13392d2abec5e073f08d646fee32ca

onurboy, 2016-06-11:
@CraigElder,
I have a problem with the zener diode issue. I am using a zener diode with a 10v 1000uf capacitor but I have noticed that when I want to use external usb adapter it does not power up the board. If I unplug the zener diode/capacitor thing on the servo rail then external usb works. I have noticed that the LED on the external usb and my dragonlink receiver require too much current together and when I disable the dragonlink receiver then external usb works again (but this time zener plugged).

Do you have any idea?
Thank you

Onur.

hamishwillee, 2016-06-13:
@onurboy Propose that you raise this discussion on the discussion boards rather than adding a technical question to the end of a closed issue.