ArduPilot/ardupilot_wiki #415: Documentation improvement for Ardurover

HuguesD, 2016-04-27:
Hello,
I find the new Ardurover documentation not as clear as it was before, especially about the first time setup, in the wiring between the flight controller servo (RC) outputs and the rover.
I suggest to add the following information on the page "quick start",  wiring, just after the current part that says that servo 1 is steering, servo 3 is throttle. To cover the case of rovers that are skid steer vehicles:
_The skid-steer parameters are used to configure vehicles that have fixed wheels and steer like tank tracks 
(do not use servos to steer the wheels but rather use differential speed between the left and right wheels)
The parameters are SKID_STEER_OUT and SKID_STEER_IN. When enabled, flight controller's ouput RC1 is used for
 the left track control, and ouput RC3 is used for right track control._

gmorph, 2016-04-27:
That seems fine HughesD.  Are you able to make the changes and submit a pull request?
Thanks, Grant.

HuguesD, 2016-04-30:
Hi Gmorph, I could do the changes but how does it work ?

HuguesD, 2016-05-01:
I tried the "pull request" tab but it seems to be for source code comparison between different versions? I do not see how that can be used to edit the doc wiki ?
Is there some interface and login/pwd for editing the wiki doc on Ardurover ?

gmorph, 2016-05-01:
Hi HughesD.  The wiki is automatically generated from source code.  Have a read through this
http://ardupilot.org/ardupilot/docs/common-wiki_editing_guide.html
and let me know if you think you can help.  It would be great if you could.
Thanks, Grant.

HuguesD, 2016-05-03:
I guess this thread can be closed since a pull request is ongoing ?

hamishwillee, 2016-05-03:
Typically you add a link to the PR. Associated issues then get closed when the PR is closed.

HuguesD, 2016-05-21:
The information is still not adapted in the wiki. When shall it be done ?
Is there a backup copy of the old wiki ? I'd like that to be able to move forward until the current wiki's content is usable. Thx

hamishwillee, 2016-05-22:
@HuguesD - I am confused - are you saying you submitted a PR that hasn't propagated into the output? If so, can you link to the page you expect updated?

The "old" wordpress wiki is gone AFAIK, but in any case any changes to this content would have been made a long time ago ... so wordpress would match this content.

HuguesD, 2016-05-22:
Hello Hamish,
I did a PR and explained in my first post the requested change. I understood my PR had to be validated by someone in charge and obviously the change is not done yet (on this page : http://ardupilot.org/rover/docs/common-pixhawk-wiring-and-quick-start.html
)
Other very useful info and schematics which were available on the previous wiki is now missing for setting up a Rover (about the wiring), as discussed here : http://discuss.ardupilot.org/t/ardurover-setup-documentation/9167
Maybe this archived page should be reactivated : http://ardupilot.org/rover/docs/rover-apm2-setup.html

hamishwillee, 2016-05-22:
@HuguesD Thanks for clarification. We won't be reactivating the old page, though I hope any issues with the wiring diagram can be "recaptured".

@gmorph Can you please make the needed updates to #424 and merge (as previously discussed) so we can close this?

closed 2016-11-15 by davidbuzz