ArduPilot/ardupilot_wiki #417: Downloading and Analyzing Data Logs in Mission Planner

CraigElder, 2016-04-30:
Parsing of tables is broken in the second half of this page.

jpkh, 2016-05-07:
you mean this: http://ardupilot.org/planner/docs/common-downloading-and-analyzing-data-logs-in-mission-planner.html
Can you post full url with link to internal headers on the area where you found this.

hamishwillee, 2016-05-07:
That's the one - search on "Compass"

davidbuzz, 2016-11-15:
fixed with commit: https://github.com/ArduPilot/ardupilot_wiki/commit/66f87ab6f42d1b48af5e39902e38cfbd10c97575   which turns all thenon-rendering html table artifacts into proper rendering tables.

closed 2016-11-15 by davidbuzz

hamishwillee, 2016-11-15:
@davidbuzz Very nice. FMI only, How did you generate the proper tables from the "html"?

davidbuzz, 2016-11-15:
@hamishwillee  - I used a two stage process: 
1 - past it into here: https://domchristie.github.io/to-markdown/  ( with the github tickbox ticked )
2 - manually fix everything that's broken, which is everything.

billbonney, 2016-11-15:
@davidbuzz nice, was marvelling at how bad that page had got on formatting just the other day (while trying to glean some useful info)