ArduPilot/ardupilot_wiki #418: explanation required for SITL locations.txt file

HuguesD, 2016-04-30:
Hello,
The ardupilot/Tools/autotest/locations.txt file allows to define starting home positions for the vehicles in SITL. A default locations.txt file exists with different locations. However there are no explanations to be found about what are the last numbers on each line (the first two numbers seem to be GPS coordinates) ?

WickedShell, 2016-04-30:
Altitude and starting heading, but it should be better documented, you're
right.

IIRC the order is Lat,Lng,Alt,Heading where alt is MSL and in meters, and
heading is degrees

HuguesD, 2016-04-30:
Thx, very clear now. Should be added to this doc page:http://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html#using-sitl-for-ardupilot-testing

magicrub, 2016-05-01:
Pending change to the wiki
https://github.com/ArduPilot/ardupilot_wiki/pull/422

peterbarker, 2016-11-15:
The referenced ticket has already been closed.

I've just pushed a change to sim_vehicle.py to ignore (shell-style) comments in locations.txt, and added a comment at the top of locations.txt giving the format.

closed 2016-11-15 by peterbarker