ArduPilot/ardupilot_wiki #419: Forwarding SITIL's Mavlink to a com port on same machine

HuguesD, 2016-04-30:
Hello,
I 'd like to forward mavlink messages from a running SITL/MAvproxy on Windows to a com port on the same machine on which a 3DR radio is connected. On the other side of the radio link is a real antenna tracker. The objective is to feed the real antenna tracker with simulated copter positions, in order to test the tracker.
Forwarding SITL/MavProxy messages to a UDP port is documented and works. Doing the same to a com port is not documented and all I have tried fails (for example running the MavProxy command : "output add com16"). It fails because SITL/MavProxy see that the com port is already in use by another device (the 3DR radio).
Is there any known solution to do this or is it impossible due to the nature of a com port in Windows ? (can only be used by one "master")

HuguesD, 2016-04-30:
Here is the error that I get trying to output om com16:

Hugues@Hugues-PC ~/ardupilot/ArduCopter
$ sim_vehicle.sh -j4 -L Brussels --console --map --out=com16
~/ardupilot/ArduCopter ~/ardupilot/ArduCopter
Building sitl
Checking modules
// BUILDROOT=/tmp/ArduCopter.build HAL_BOARD=HAL_BOARD_SITL HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_NONE TOOLCHAIN=NATIVE EXTRAFLAGS=-DGIT_VERSION="4bf3ec0e"
~/ardupilot/ArduCopter
Starting up at Brussels : 50.808250,4.392532,21,360
Using defaults from ../Tools/autotest/copter_params.parm
Starting ardupilot : /home/Hugues/ardupilot/ArduCopter/ArduCopter.elf -S -I0 --home 50.808250,4.392532,21,360 --model + --speedup=1 --defaults ../Tools/autotest/copter_params.parm
Window access not found, logging to /tmp/ardupilot.log
../Tools/autotest/run_in_terminal_window.sh: line 28:  4196 Terminated              $cmd $* &> $filename < /dev/null

HuguesD, 2016-04-30:
I tried different syntaxes with same exit : "com16", com16:, etc

HuguesD, 2016-04-30:
I also tried the non command line "output add com16" command, which gives also an error:

STABILIZE> output add com16
STABILIZE> Adding output com16
Exception in thread main_loop:
Traceback (most recent call last):
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.py
z\threading", line 810, in __bootstrap_inner
  File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.py
z\threading", line 763, in run
  File "<string>", line 756, in main_loop
TypeError: argument must be an int, or have a fileno() method.

gmorph, 2016-05-01:
This does work.  If your using cygwin the ports aren't COM.  You should use /dev/ttyS15 (this is com16 - they start at 0).
sim_vehicle.sh -j4 -L Brussels -A --uartC=uart:/dev/ttyS15 --console --map 
I haven't tested this so let us know how you go.  You may need to specify the baudrate as well.
Thanks, Grant.

HuguesD, 2016-05-02:
I have tried the /dev/ttty thing.
Here is my /dev directory which shows my 3DR radio connected on ttyS29:
crw-rw-rw-  1 Hugues None   5,   0 May  2 19:52 tty
crw-rw-rw-  1 Hugues None 117,   0 May  2 19:52 ttyS0
crw-rw-rw-  1 Hugues None 117,  29 May  2 19:52 ttyS29
crw-rw-rw-  1 Hugues None   1,   9 May  2 19:52 urandom
crw-rw-rw-  1 Hugues None  13, 255 May  2 19:52 windows
crw-rw-rw-  1 Hugues None   1,   5 May  2 19:52 zero

Then I start : sim_vehicle.sh -j4 -L Brussels -A --uartC=uart:/dev/ttyS29 --console --map 
Then I get indeed the Mavproxy window, map and console.
In the Mavproxy window I check the outputs and do not see the tty29 (Or COM):
STABILIZE>
STABILIZE> output
STABILIZE> 2 outputs
0: 127.0.0.1:14550
1: 127.0.0.1:14551

=>No mention of a TTY29 added output...no good.

The antenna tracker is on, armed, in auto mode and has the other side of the tty29 radio on but does not move, as if it was not connected to a vehicle (which is probably the case).

With the other pair of radio I can connect with mission planner on the Antenna tracker but nothing special to mention (I did a manual servo motion test, that works fine). I tried scan, auto mode, nothing moves.

I believe the --uartC=uart:/dev/ttyS19 does not add an output. I think it is used for HIL where you tell SITL that it must take its GPS input from a real GPS that would be connected on ttyS19.

So that does seem to be the trick to add a mavproxy output to a 3DR radio connected on a COM port, unfortunately. 
Any other idea ?

gmorph, 2016-05-03:
Hi Hughes.  The "-A" flag means the data won't go via mavproxy.  It will go straight from SITL out the comm port.  This means you won't see it as an output in MAVProxy.
Note I've never done this (I'm a Linux guy) but it does work as we have had people do it.
So, as a test can you instead of having your radio on your antenna tracker instead connect it to a computer with MAVProxy and see if you can receive the data from the SITL copter?
Thanks, Grant.

HuguesD, 2016-05-03:
When you say "connect it to a computer with Mavproxy", do you mean a usb cable from the PIxhawk on the antenna tracker to the PC running SITL ?

gmorph, 2016-05-04:
This is what I'm thinking.  You have a SITL vehicle which is akin to a real physical vehicle.  It is outputting its mavlink   data via the 3DR Radio thanks to the  -A --uartC=uart:/dev/ttyS29  just like a real vehicle would.  Now you would normally have a GCS with a radio of its own receiving that data.  So why not try that.  Instead of having the data going to your Antenna tracker have it go to a GCS with its own 3DR Radio.  You can use the same computer.  Just plug in a second 3DR Radio and have your GCS of choice connect to that comm port.  Or a different computer if you prefer.
Make sense?
Thanks, Grant

HuguesD, 2016-05-04:
[image]
Grant, I understand the setup you propose which is in fact putting a "GCS in the middle" (on the right is the vehicle with a 3DR link to the GCS, on the left is the antenna tracker connected to the GCS by USB or radio also). This is not what I'm trying to do because the "GCS in the middle" does not allow me to use the tracker for receiving long range telemetry (I do not use the tracker for video, only for maintaining a long range telemetry/mavlink link between the GCS and the vehicle).
My wished setup is a "ANtenna tracker" in the middle (on the right is the vehicle with a 3DR link to the antenna tracker, on the left is the GCS linked to the antenna tracker with a second pair of 3DR radios - see attached picture).
My setup works when using all real elements (real drone, real antenna tracker and real GCS with mission planner) like in then picture. 
The issue is that it is difficult to test and tune, therefore I just wanted to replace the "real vehicle" by a SITL vehicle running on the same PC then the GCS (could be running on another one if need be, it doesn't matter).
My issue is to link the telemetry/mavlink between the SITL and the Pixhawk of the antenna tracker (via telem1 port). The  -A --uartC=uart:/dev/ttyS29 does not do it in my last test. Maybe I'm still doing something wrong or forgot to configure something.
Side note : I do not think anyone else has already done this setup as most others have tried the GCS in the middle setup where it is the GCS sending the servo orders to the antenna tracker. It is different in my case where antenna tracker has to receive full mavlink packets from SITL, not just PWM RC commands sent from a GCS to an arduino board+servos on the tracker. Honestly, I looked through google, forums and have never found anybody make it work the way I'd like to. By the way , I have seen others asking how to setup in the same way as what I describe, so I'm not alone in the dark lol.

I will try again with the -A command above that probably is sending something out the COM port, but I wonder if the issue does not reside in the code of antenna tracker that would not read from the telemetry port of Pixhawk to interpret the incoming mavlink flow ? Who's the "lead" developper of the antenna tracker that I could ask the question to because it is outside my skills to interpret the code ?

HuguesD, 2016-05-05:
Solution was found and given by Sebastian on diydrones : 
http://diydrones.com/forum/topics/figuring-out-the-antenna-tracker?commentId=705844%3AComment%3A2250000&xg_source=msg_com_forum

Antenna tracker/SITL mavlink wiki is wrong and when I'll have figured out completely how it works, I'll post a wiki modification request.

hamishwillee, 2016-05-17:
(or you can just update the page yourself as a PR! http://ardupilot.org/ardupilot/docs/common-wiki_editing_guide.html)

davidbuzz, 2016-11-15:
90% of this issue is just debugging someones mavproxy and SITL setup. no improvements to the wiki have been offered at this stage, so I'm closing this ticket.     If you'd like to have better documentation of the solution that was "found", please document your method, and create a new issue, and we'll update the docs with your new method. thanks.

closed 2016-11-15 by davidbuzz

RMFPV, 2020-05-31:
What is the solution?
Please post here for Win.
Thnx