ArduPilot/ardupilot_wiki #439: Package "gflags_catkin" is required in page "Using ROS/Gazebo Simulator with SITL"

glody, 2016-05-10:
In page "http://ardupilot.org/dev/docs/using-rosgazebo-simulator-with-sitl.html", error "glog_catkin: Cannot locate rosdep definition for [gflags_catkin]" shows while runing "rosdep install --from-paths src --ignore-src --rosdistro indigo -y".

Pull request had created to fix this issue: https://github.com/ArduPilot/ardupilot_wiki/pull/438

Error message:
$ rosdep install --from-paths src --ignore-src --rosdistro indigo -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
glog_catkin: Cannot locate rosdep definition for [gflags_catkin]

kiti1013, 2016-10-14:
Hello, do you solve this problem？

khancyr, 2017-02-16:
Can be close

closed 2017-10-15 by auturgy