ArduPilot/ardupilot_wiki #453: pixracer quick start?

warmwater, 2016-05-27:
It will be nice to have pixracer quick start or detailed instruction

hamishwillee, 2016-05-28:
Yes it would - at the moment all we've got is the [pixracer overview](http://ardupilot.org/copter/docs/common-pixracer-overview.html). If you have any expertise in creating something similar to the [Pixhawk Quickstart](http://ardupilot.org/copter/docs/common-pixhawk-wiring-and-quick-start.html) we'd be happy to take it :-)

@rmackay9 Anyone else you might suggest we could get on board to create a quickstart?

davidbuzz, 2016-11-15:
No movement on this suggestion  for ~6 months, closing it, but I encourage the original submitter to rais a PR for the wiki with this suggestion.

closed 2016-11-15 by davidbuzz