ArduPilot/ardupilot_wiki #455: Mission Planner: Download Logs does not rename filenames according to GPS date since 1.3.38

t2000, 2016-05-29:
Hello,

since I updated MissionPlanner to version 1.3.38 I can still download my log files from my APM. However, the filenames are 1 1-1-1970 1-00-00 AM.bin and not like the GPS date+time within the log file. Before in version 1.3.37 MissionPlanner looked INTO the logfiles, extracted the GPS date+time and renamed the logfile accordingly.

Regards,

Stefan

hamishwillee, 2016-05-29:
@t2000 This is a place for wiki issues, not ardupilot or mission planner bugs. Can you please repost against whichever of those repositories you think this belongs in (possibly Mission Planner repo)

closed 2016-05-29 by hamishwillee

skinnyb8, 2016-05-30:
Dear Sir;  after a year and two smoked APM 2.8s I finally got my system up and running and actually  worked!!! Then Mission Planner “Automatically” Upgraded and everything quit…. That was my first information that the system was removed from the internet (for the 2.8  8 bit APM).

 

Since then I’ve spent countless Hours chasing and searching for 
WHERE SHOULD I address my issues and continue with WHAT I’ve 
GOT rather that spend several More $100 trying to START ALL OVER….

 

WE need HELP! Not being told to “Go Away and Leave Ya’ll Alone”…

 

I fully understand the effort to move ahead to 32 bit or whatever but we shouldn’t be Stopped…

 

Kenneth D Bryant

 

Closed #455 <https://github.com/ArduPilot/ardupilot_wiki/issues/455> .

hamishwillee, 2016-05-30:
@skinnyb8 The right place to start asking general technical support questions is on the discussion boards - probably the [Mission Planner discussion board](http://discuss.ardupilot.org/c/ground-control-software/mission-planner). 

They should be able to work out why your firmware updated - it should not have done so if you're on old hardware, and the best way to revert. 

This place is for asking questions about wiki content and suggesting improvement.

magicrub, 2016-05-30:
Discuss.ardupilot.org is the main portal for forum help.  Have you posted
there yet?