ArduPilot/ardupilot_wiki #457: quadplane ESC calibration section missing

haogodent2, 2016-05-31:
following section was recently omitted after reorg.:

> 
> ESC calibration
> Most models of PWM based ESC need to be calibrated to ensure that all the ESCs respond to the same input with the same speed. To calibrate them they need to receive maximum PWM input when initially powered on, then receive minimum PWM input when they have beeped to indicate that the maximum has registered.
> 
> The quadplane code doesn’t have a dedicated ESC calibration feature yet, but you can use the following procedure to calibrate until that is available:
> 
> remove your propellers for safety
> power up just the flight board and not your motors. If you don’t have the ability to isolate power to the ESCs when on battery power then power up your flight board on USB power
> set both the parameters Q_M_SPIN_ARMED and Q_THR_MID to 1000. This sets the PWM output when armed at zero throttle to full power
> set the safety switch off to activate the outputs
> arm your aircraft. The PWM output on all quad motors will now climb to maximum.
> add power to your ESCs by connecting the battery
> wait for the ESCs to beep to indicate they have registered the maximum PWM
> disarm your aircraft. The ESCs should beep again indicating they have registered minimum PWM
> Now set the Q_M_SPIN_ARMED and Q_THR_MID parameters back to the correct values. A value of 50 for Q_M_SPIN_ARMED is a reasonable starting point. For Q_THR_MID a value of between 500 and 600 is good depending on the power of your motors

magicrub, 2016-05-31:
@tridge recently did some work on quadplane docs where he split up a huge page with multiple sections into several small pages. This section (ESC cal) did not get it's own page and was lost in the re-org.

hamishwillee, 2016-05-31:
@magicrub Deliberate or just an omission?

magicrub, 2016-05-31:
It was an accident. I've talked to him about it and he said he's working on it

gmorph, 2016-11-15:
The ESC calibration page can be found here:
http://ardupilot.org/plane/docs/quadplane-esc-calibration.html

closed 2016-11-15 by gmorph