ArduPilot/ardupilot_wiki #458: Flymaple based content should be removed from the docs

hamishwillee, 2016-05-31:
Flymaple support was recently removed from ArduPilot

> flymaple removal (#4191)

Any content on the topic should be removed from the docs

magicrub, 2016-06-17:
I think this has been fixe dalready

hamishwillee, 2016-06-19:
No. I didn't do it and haven't seen anyone else doing it. There are at least two hits here: http://ardupilot.org/dev/search.html?q=flymaple&check_keywords=yes&area=default

We should also update the history (http://ardupilot.org/copter/docs/common-history-of-ardupilot.html?highlight=flymaple) to show removal of support.

davidbuzz, 2016-11-15:
I've removed references to the flymaple except for https://github.com/ArduPilot/ardupilot_wiki/blob/master/dev/source/docs/building-apm-for-flymaple.rst  , as it explicitly mentions the board is no longer supported.     I've also added the "removal of support" to the history document.   issue can now be closed.

closed 2016-11-15 by davidbuzz