ArduPilot/ardupilot_wiki #461: Adding a new MAVLink Message (Code Overview)

uncleyoung, 2016-06-06:
[Adding a new MAVLink Message (Code Overview)](http://ardupilot.org/dev/docs/code-overview-adding-a-new-mavlink-message.html) 
There is no generate.sh script in libraries/GCS_MAVLink folder  in latest version. I am confused about how to play around with this tutorial. Maybe this tutorial should be updated?
It seems mavlink src will be generated automatically when built now( after all I built successfully ), but it still confuses me since I just want have a look at the mavlink src code/content.

magicrub, 2016-06-06:
Thanks! Fixed via
https://github.com/ArduPilot/ardupilot_wiki/commit/7c06f57d901a958d72dca8492df08a50e857e7e4

hamishwillee, 2016-06-06:
@magicrub - Did your changes catch any other stuff in #354 (some good notes on changes needed that I have not yet have time to look at properly )

magicrub, 2016-06-06:
I fixed most items from this and that other issue. This issue is completed and a duplicate.  Closing.

closed 2016-06-06 by magicrub