ArduPilot/ardupilot_wiki #464: Lidar-lite doc inconsistency

tuloski, 2016-06-07:
In the documentation of the Lidar-Lite (common/source/docs/common-rangefinder-lidarlite.rst)
it is written that RNGFND_TYPE should be 5, but later it's written it should be 3 (APM2) or 4 (Pixhawk).
Which one is the correct information?

hamishwillee, 2016-06-07:
@rmackay9 This appears to be a legitimate flaw in the docs

Based on the [parameter docs](http://ardupilot.org/copter/docs/parameters.html?highlight=rngfnd_type#rngfnd-type-rangefinder-type) I believe you set like this:

3:	APM2 and I2C connection
4:	PX4 and I2C connection
5:	PX4 and PWM connection

The fix that would be applied would be to add PWM vs I2C information in [this section](http://ardupilot.org/copter/docs/common-rangefinder-lidarlite.html#setup-in-mission-planner): 

> To configure Copter, Plane or Rover to use the LIDAR-Lite, please first connect with the Mission Planner and then open the Config/Tuning | Full Parameter List page and set: RNGFND_TYPE to “3” if using an APM2, or “4” if using a PX4 or Pixhawk.

Can you confirm I've got this right?

rmackay9, 2016-06-08:
yes, that's right.  thanks very much!

hamishwillee, 2016-06-08:
[All fixed](http://ardupilot.org/copter/docs/common-rangefinder-lidarlite.html#setup-in-mission-planner) - thanks @rmackay9 and @tuloski

closed 2016-06-08 by hamishwillee