ArduPilot/ardupilot_wiki #468: Add huge banner stating last firmware version that works on APM 2.x

magicrub, 2016-06-09:
Developers get asked all the time about running the latest ArduPilot features on APM 2.6 hardware. It would be nice to not repeat ourselves explaining that an 8bit cpu with extremely limited RAM and program space can't handle it and that the recommended hardware is Pixhawk.

A page dedicated to this would be nice, explaining why we moved up in hardware, where to get it (link to other wiki page)

hamishwillee, 2016-06-13:
I'm not sure how much more clear we can be: http://ardupilot.org/copter/docs/common-apm25-and-26-overview.html?highlight=apm

I guess we could create a page at this level http://ardupilot.org/copter/docs/assembly-instructions.html but I suspect the problem is that people just don't read.

magicrub, 2016-06-13:
searching "apm 2.5" or "apm 2.x" or "apm 2.6" don't bring up anything.

hamishwillee, 2016-06-14:
True. I believe that the indexing code will split on non-word characters, and also possibly omit short words from the index. So APM and APM 2 work fine, but APM 2.5 does not. While that is reasonable I think it is a problem that we can't even use glossary or definition markup to force a keyword like APM 2.5 into the index. I added to [this bug with that feedback](https://github.com/sphinx-doc/sphinx/issues/1918#issuecomment-225745290).

I am not sure how to fix this. The best fix would be to extend the search routine but I do not have time to do/investigate that if unpaid.

The big page at http://ardupilot.org/copter/docs/assembly-instructions.html is another idea, but that is a bit of a patch, not a solution.

Thoughts?

Thoughts?

davidbuzz, 2016-11-15:
I've followed the logical progression through the wifi for a first-time user who's trying to get started, and made it easier to find at least one link to APM2 stuff, which will them tell them it's been retired/obsoleted. Home->Copter->First Time Setup->Assembly->APM2  seems logical to me see: https://github.com/ArduPilot/ardupilot_wiki/commit/2cb7a70b573808b0d8cd51dea98fa729dca16a4d      I think that should resolve this issue.

closed 2016-11-15 by davidbuzz

hamishwillee, 2016-11-15:
Yes, this works for me. I'd still be keen to get glossary terms added to the search index though.