ArduPilot/ardupilot_wiki #471: Building for Pixhawk/PX4 using Eclipse on Windows, Java and Make issues

jdbarnes, 2016-06-10:
[Building for Pixhawk/PX4 using Eclipse on Windows](http://ardupilot.org/dev/docs/editing-the-code-with-eclipse.html) - I am trying to install and run Eclipse to compile firmware for Pixhawk. I have tried multiple options that are outlined on the ardupilot developer site but can't get any of them to work. I am new to setting up an IDE and compiler for C/C++ and this is proving to be very challenging. If I could get some feedback on how to get Eclipse (or some other method, perhaps Notepad++ and an associated compiler?) I would really appreciate it. Below is what I have tried:

For Eclipse:
Installed the toolchain that is linked (px4_toolchain_installer_v14_win.exe) but when I tried to run eclipse it returns "exith code=13" and does not run. I have tried the possible fixes of installing java (jdk1.8.0_91) and including the -vm command in the eclipse.ini file (-vm
C:\Program Files\Java\jdk1.8.0_91\bin\javaw.exe) and also setting the environment variable PATH to C:\Program Files\Java\jdk1.8.0_91\bin. None of this worked.

I suspect that I have installed a newer version of jdk which I believe is 64 bit and the toolchain was written for 32 bit? I don't know if trying to install an older version of JDK would solve the problem.

When I encountered this I tried installing the latest version of Eclipse and it works with the current JDK but when I try to compile "px4-v2" with the cloned Ardupilot repository Eclipse displays the error "Program "make" not found in PATH." I don't understand how to fix that either.

Please help me start developing for Pixhawk, thank you!

hamishwillee, 2016-06-13:
The instructions worked earlier in the year when I tested them. Unfortunately I don't have time to retest them now. You should ask for support on the discussion forums.

Things to try:

* Can you build without the IDE? Ie follow the instructions here: http://ardupilot.org/dev/docs/building-px4-with-make.html
* You might also want to check out #463 which shows how to set up things using QtCreator.

davidbuzz, 2016-11-15:
This looks like the issue is specific to the user who raised it.   Please ask your question/s at:   http://discuss.ardupilot.org and oncer you are sure the wiki itself is wrong, please open a new issue with a fix, and we'll merge it.

closed 2016-11-15 by davidbuzz

hamishwillee, 2016-11-15:
@davidbuzz Just as an FYI, at some point soon we should try create something about how to integrate IDEs with our new build toolchain and move this to archive.