ArduPilot/ardupilot_wiki #474: Adding a New Parameter - missing info for new library class

dricketts, 2016-06-14:
[Adding a New Parameter (Code Overview)](http://ardupilot.org/dev/docs/code-overview-adding-a-new-parameter.html) -

There seems to be a little bit of information missing from this page for adding a parameter to a new library class. In particular, I think that the following steps are missing:

1. Add ```static const struct AP_Param::GroupInfo var_info[];``` to the public section of the .h file of new library class in addition to the definition in the .cpp file.

2. Add ```k_param_my_new_lib``` to Parameters.h where ```my_new_lib``` is the first argument to the ```GOBJECT``` declaration in Parameters.cpp.

Finally, I think that in the code snippet in step 2 of "Adding a parameter to a library", the word ```PROGMEM``` should be removed.

gmorph, 2016-11-16:
Please make the necessary changes and submit a PR and we will merge them in.
Thanks, Grant.

brunoolivieri, 2019-08-12:
I believe that PR #603 fixed that. So, I believe we could close this issue now.

closed 2019-08-12 by Hwurzburg