ArduPilot/ardupilot_wiki #484: RSSI documentation Outdated

PaulHam211, 2016-07-05:
I have noticed that the documentation is outdated for sensing RSSI. I noticed this today when upgrading to the latest version from a old old version.

Here is the outdated page 
https://github.com/ArduPilot/ardupilot_wiki/blob/master/common/source/docs/common-rssi-received-signal-strength-indication.rst

I would try to correct this, but currently I am also stuck as to how it works now. There isnt a description next to the new values when looking at the params when connected to the plane.

gmorph, 2016-11-15:
Which bit is outdated exactly?  I think the page as shown still works?

PaulHam211, 2016-11-16:
Sorry, I can't actually remember what bit was outdated now. But when I tried this on my Pixhawk it confused me. I'll be setting up another Pixhawk in a few weeks so ill come back and update when I hit the issue again.

brunoolivieri, 2019-08-20:
Lots of information have been added and updated from the last yeards (Commits history:  https://github.com/ArduPilot/ardupilot_wiki/commits/master/common/source/docs/common-rssi-received-signal-strength-indication.rst )

As there is not a specific concern, I believe we should close this issue. Please if you don't agree, request, and we can reopen the issue.

closed 2019-08-20 by brunoolivieri