ArduPilot/ardupilot_wiki #485: Visiting Camera Gimbal setup page resets Cam_Trigg_Type

sandysoundy, 2016-07-06:
Anyone else have this problem?
Whenever I go: Initial Setup, Optional Hardware, Camera Gimbal page it resets my Cam Trig type to 0 (servo) when I want it to stay 1 (Relay).
Whenever I visit that Camera Gimbal setup and then try to Trigger Camera NOW from the Flight Data screen it doesn't work, so I have to change the value in Config Tuning, Full Paramater List, Cam_Trigg_Type back to 1.
EVERY TIME!
Even if I have written it to the Pixhawk prior to visiting the camera gimbal setup, it will still go back to servo type trigger. I have tried writing the parameter change and rebooting the pixhawk, and refreshing parameters but as soon as I visit that camera gimbal setup page it changes it back to 0.
Feel like I am chasing my tail with it. I have certainly lost time in the field over it trying to figure out why the camera was working but now isn't!
The reason I visit the Camera Gimbal page is to fine tune the camera trigger length. And if I want to make an adjustment there, I have to go back to the full parameters list to change the trigg type, and then test it with the trig camera NOW in Flight data screen and rinse and repeat till I am happy with how the camera is behaving. Totally boring!
For the record I am using the latest Mission Planner with latest stable FW.

hamishwillee, 2016-07-08:
@meee1 - Should MP reset the camera trigger type as described here?

@sandysoundy - Best to raise this sort of discussion on the forums: http://discuss.ardupilot.org/c/ground-control-software/mission-planner

sandysoundy, 2016-07-08:
It's a known and long-standing issue apparently. 
Thanks for the link, I thought this was the right place...

hamishwillee, 2016-07-09:
No worries. This is the right place to report issues with the documentation itself. The discussion boards are a good place for technical support and to discuss bugs before they are formally posted.

Closing this now.

closed 2016-07-09 by hamishwillee