ArduPilot/ardupilot_wiki #490: Building ArduPilot for Pixhawk/PX4 on Windows with Make

aguestguy, 2016-07-16:
When I try to build the firmware and enter "make px4", "make px4-v2", or "make px4-clean", it runs for a while before it suddenly stops and displays:

In file included from c:/GIT/ardupilot/libraries/AP_HAL/utility/sparse-endian.h:23:0,
                 from c:/GIT/ardupilot/libraries/AP_ADC/AP_ADC_ADS1115.cpp:2:
c:/GIT/ardupilot/libraries/AP_Common/missing/byteswap.h: In function 'uint16_t __bswap_16(uint16_t)':
c:/GIT/ardupilot/libraries/AP_Common/missing/byteswap.h:11:42: error: '__builtin_bswap16' was not declared in this scope
compilation terminated due to -Wfatal-errors.
make[3]: *** [libraries/AP_ADC/AP_ADC_ADS1115.cpp.o] Error 1
make[3]: Leaving directory `/c/GIT/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/c/GIT/ardupilot'
make[2]: *** [/c/GIT/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build//c/GIT/ardupilot/module.pre.o] Error 2
make[2]: Leaving directory `/c/GIT/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build'
make[1]: *** [/c/GIT/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.px4] Error 2
make[1]: Leaving directory `/c/GIT/ardupilot'
make: *** [px4-v2] Error 2

I've tried searching online, but I couldn't seem to find any relevant information about these errors.

magicrub, 2016-07-16:
You're sure you get that when using make px4-clean?

aguestguy, 2016-07-16:
er sorry actually when I use make px4-clean I get:

../mk/px4_targets.mk:27: *** You have an old PX4Firmware tree - see http://dev.ardupilot.com/wiki/git-submodules/
Checking modules

then it stops

magicrub, 2016-07-16:
and that link takes you to the wiki that shows that exact error and it says to run:
git submodule update

and after following that wiki instruction you're still having problems?

aguestguy, 2016-07-16:
oh thanks! sorry I misunderstood that part. I thought the instructions for that error meant that I didn't need to worry about it, not that I needed to manually update the submodules. When I ran "git submodule update" from the ardupilot folder nothing happened. and when I tried make px4-v2 again in the Arducopter folder I got the same error from my first post

hamishwillee, 2016-07-18:
Perhaps doing a submodule init first:
```
git submodule init
git submodule update
```
OR
```
git submodule update --init --recursive
```
The second line is suggested on Windows: http://ardupilot.org/dev/docs/building-px4-with-make.html

Clean everything else up first, and re-sync to the head.

aguestguy, 2016-07-18:
I ran `make clean`, re-synced to the head, and tried both but I still get the same error when I enter `make px4-v2.`
although `git submodule init` and `git submodule update` seem to work without any problem, when I do `git submodule update --init --recursive` , this is what shows up:
```
$ git submodule update --init --recursive
Submodule 'modules/PX4Firmware' () registered for path 'modules/PX4Firmware'
Submodule 'modules/PX4NuttX' () registered for path 'modules/PX4NuttX'
Submodule 'modules/gbenchmark' () registered for path 'modules/gbenchmark'
Submodule 'gtest' () registered for path 'modules/gtest'
Submodule 'modules/mavlink' () registered for path 'modules/mavlink'
Submodule 'modules/uavcan' () registered for path 'modules/uavcan'
Submodule 'modules/waf' () registered for path 'modules/waf'
Submodule 'Tools/gencpp' (git://github.com/ArduPilot/PX4Firmware.git) registered for path 'Tools/gencpp'
Submodule 'Tools/genmsg' (git://github.com/ArduPilot/PX4Firmware.git) registered for path 'Tools/genmsg'
Submodule 'mavlink/include/mavlink/v1.0' (git://github.com/ArduPilot/PX4Firmware.git) registered for path 'mavlink/include/mavlink/v1.0'
Submodule 'src/lib/DriverFramework' (git://github.com/ArduPilot/PX4Firmware.git) registered for path 'src/lib/DriverFramework'
Submodule 'src/lib/ecl' (git://github.com/ArduPilot/PX4Firmware.git) registered for path 'src/lib/ecl'
Submodule 'src/lib/matrix' (git://github.com/ArduPilot/PX4Firmware.git) registered for path 'src/lib/matrix'
Submodule 'src/modules/uavcan/libuavcan' (git://github.com/ArduPilot/PX4Firmware.git) registered for path 'src/modules/uavcan/libuavcan'
Submodule 'cmake_hexagon' (https://github.com/PX4/DriverFramework.git) registered for path 'cmake/cmake_hexagon'
Submodule 'dspal' (https://github.com/PX4/DriverFramework.git) registered for path 'dspal'
Submodule 'test/dspal_tester/cmake_hexagon' (https://github.com/PX4/dspal) registered for path 'test/dspal_tester/cmake_hexagon'
Submodule 'matrix' (https://github.com/PX4/ecl.git) registered for path 'matrix'
Submodule 'dsdl' (git://github.com/UAVCAN/libuavcan.git) registered for path 'dsdl'
Submodule 'libuavcan/dsdl_compiler/pyuavcan' (git://github.com/UAVCAN/libuavcan.git) registered for path 'libuavcan/dsdl_compiler/pyuavcan'
Submodule 'dsdl' (https://github.com/UAVCAN/pyuavcan) registered for path 'dsdl'
fatal: Could not chdir to '../../../../../../../../../../../../../../../modules/PX4Firmware/src/modules/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl': No such file or directory
Unable to find current revision in submodule path 'dsdl'
Failed to recurse into submodule path 'libuavcan/dsdl_compiler/pyuavcan'
Failed to recurse into submodule path 'src/modules/uavcan/libuavcan'
Submodule 'dsdl' (git://github.com/ArduPilot/uavcan.git) registered for path 'dsdl'
Submodule 'libuavcan/dsdl_compiler/pyuavcan' (git://github.com/ArduPilot/uavcan.git) registered for path 'libuavcan/dsdl_compiler/pyuavcan'
Submodule 'dsdl' (https://github.com/UAVCAN/pyuavcan) registered for path 'dsdl'
Failed to recurse into submodule path 'modules/PX4Firmware'
```

magicrub, 2016-07-18:
The developers are having issues with master for make right now on all
hardware but it will be resolved soon. Meanwhile, try waf.

magicrub, 2016-07-18:
quote from @OXINARF on gitter:

> He is probably using an old compiler, it is the same as #4486, you can point him to https://github.com/ArduPilot/ardupilot/issues/4486#issuecomment-233128236

bugobliterator, 2016-07-18:
@aguestguy @magicrub definitely a compiler version issue make sure you have arm-none-eabi-gcc --version >4.9

bugobliterator, 2016-07-18:
@aguestguy @magicrub it seems the current tools installer installs 4.6 version of arm-gcc as per @rmackay9 . So we need to update that.

aguestguy, 2016-07-18:
@magicrub @bugobliterator Thanks, I copied what EwingKang did from #4486 and replaced the arm-none-eabi, bin, lib, and share folders in c:\px4\toolchain\ with the ones from gcc-arm-none-eabi-4_9-2015q3-20150921-win32. Although it compiled for longer it still ended with this error: 
```
Pruning ROMFS files.
ROMFS:   romfs.img
/bin/sh: genromfs: command not found
make[2]: *** [romfs.img] Error 127
make[2]: Leaving directory `/c/GIT/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build'
make[1]: *** [/c/GIT/ardupilot/modules/PX4Firmware/Build/px4fmu-v2_APM.build/firmware.px4] Error 2
make[1]: Leaving directory `/c/GIT/ardupilot'
make: *** [px4-v2] Error 2
```
I couldn't find anything about this error occurring on windows, would either of you know how to resolve this?

rmackay9, 2016-07-19:
This compile issue is fixed in master now so closing.  We still need to upgrade the windows compiler (borrowed from the px4 team) to be 4.9 but that can be a separate issue.

closed 2016-07-19 by rmackay9

aguestguy, 2016-07-19:
@rmackay9 sorry, but you wouldn't happen to know how to resolve that issue with romfs would you?

bugobliterator, 2016-07-19:
@aguestguy you can try building genromfs binary on windows, add the binary to environment path and see if that works

aguestguy, 2016-07-22:
@bugobliterator how do I do that? I tried following the instructions on https://pixhawk.org/dev/nuttx/genromfs but I get this result when i try the third step:
```
gcc -O2 -Wall -DVERSION=\"0.5.7\" genromfs.c -c -o genromfs.o
genromfs.c:66:21: fatal error: fnmatch.h: No such file or directory
 #include <fnmatch.h>
                     ^
compilation terminated.
Makefile:28: recipe for target 'genromfs.o' failed
make: *** [genromfs.o] Error 1

```

I found this thread: https://groups.google.com/forum/#!msg/px4users/0HNsQHDpZRM/6HoLbURF20AJ but changing the access rights didn't fix anything

LarsGardien, 2017-04-22:
Just wanted to add to this closed thread since I have found another solution to this problem that worked for me. Simply move the entire repository to the root of your disk (Or any short path). Since during the submodule creation process the max filename length of you OS might be exceeded.

soheil2017, 2017-06-15:
when I run 
cd Firmware
make px4fmu-v2_default
I receive following error. who does know and solution of this error?

 couldn't find python module jinja2:

  for debian systems try:

        sudo apt-get install python-jinja2

  or for all other OSs/debian:

        pip install jinja2

   (missing:  PY_JINJA2)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.8/Modules/FindPackageHandleStandardArgs.cmake:377 (_FPHSA_FAILURE_MESSAGE)
  cmake/common/px4_base.cmake:622 (find_package_handle_standard_args)
  CMakeLists.txt:293 (px4_find_python_module)

-- Configuring incomplete, errors occurred!
See also "C:/px4/Firmware/build_px4fmu-v2_default/CMakeFiles/CMakeOutput.log".
rm: cannot remove directory `/c/px4/Firmware/build_px4fmu-v2_default': Permission denied
make: *** [px4fmu-v2_default] Error 1

OXINARF, 2017-06-16:
@soheil2017 Not sure where you got those instructions from, but they aren't for ArduPilot. Also, this isn't the place for development questions, for that use the forum (http://discuss.ardupilot.org) or Gitter chat (http://gitter.im/ArduPilot/ardupilot).

soheil2017, 2017-06-16:
OK thanks

soheil2017, 2017-06-16:
I wanna build ArduPilot for Pixhawk/PX4 on Windows with Make. is it possible introduce me a valid reference to do it. i am going to do it from zero step by step and i had problem, i will ask you. unfortunately I tested many ways. but i got error in running ## make px4fmu-v2_default . my target is generating c++ code from matlab simulink and upload it in pixhawk hardware. 

Thanks

magicrub, 2017-06-16:
@soheil2017  this isn't the place for development questions, for that use the forum (http://discuss.ardupilot.org) or Gitter chat (http://gitter.im/ArduPilot/ardupilot).