ArduPilot/ardupilot_wiki #500: Complete Parameter List edit on github link broken

ctech4285, 2016-07-27:
[Complete Parameter List](http://ardupilot.org/copter/docs/parameters.html) 

The edit on github link on this page resolves to 404 error  https://github.com/ArduPilot/ardupilot_wiki/blob/master/copter/source/docs/parameters.rst

hamishwillee, 2016-07-27:
Thanks @ctech4285,

Arggg - I thought I'd killed that link in the theme :-(. It shouldn't appear at all because this page is generated from the ardupilot source code. 

Assigned to me. I'll look at some point soon hopefully.

gmorph, 2016-11-15:
I've confirmed this is still an issue.

auturgy, 2017-10-15:
@OlivierJB @hamishwillee can this be fixed?  Should be simple for someone with access to and familiarity with the theme.

hamishwillee, 2017-10-15:
Yes, but requires a little research (which I do not have time for right now).

The edit link is dynamically created based on page name/url to point to the right place in github. The reason it fails is that the source for this page *is not in github* - the page is dynamically generated by a script at build time from source.

The solution is to hide the edit link on the page altogether. I'm no longer sure how to do that. BUT I did kill the edit link on the home page, so there must be some theme code to make this happen already. Probably just needs to also include the parameters page.

Hwurzburg, 2019-11-02:
@hamishwillee any progress on this??

hamishwillee, 2019-11-03:
See link above. I hope you can test it for me?

closed 2019-11-03 by hamishwillee