ArduPilot/ardupilot_wiki #501: Parameter List BRD_CAN_ENABLE needs updating

ctech4285, 2016-07-27:
http://ardupilot.org/copter/docs/parameters.html#brd-can-enable-enable-use-of-uavcan-devices
BRD_CAN_ENABLE: Enable use of UAVCAN devices should be:

 *  0 - UAVCAN disabled.
 *  1 - Enabled support for UAVCAN actuators and sensors.
 *  2 - Enabled support for dynamic node ID allocation and firmware update.
 *  3 - Sets the motor control outputs to UAVCAN and enables support for dynamic node ID allocation and firmware update.
 *
https://github.com/PX4/Firmware/blob/e1868216927ee3a33a98b3b9d309dfc13a506703/src/modules/uavcan/uavcan_params.c#L41-L57

hamishwillee, 2016-07-27:
Hi @ctech4285 
The parameter documentation is actually generated dynamically from source. Can you please raise this issue in the ardupilot (source) repo here https://github.com/ArduPilot/ardupilot ?

ctech4285, 2016-07-28:
Issue raised, thank you Hamish

gmorph, 2016-11-15:
This isn't actually an issue as the code for 2 and 3 isn't currently in master but there is a PR tracking it so we can close this.

closed 2016-11-15 by gmorph