ArduPilot/ardupilot_wiki #504: Suggestions for "Adding Parameters" Dev Wiki Page

olliw42, 2016-08-01:
Below please find a suggestion for improving the developers wiki page on "Adding Parameters" (http://copter.ardupilot.com/dev/docs/code-overview-adding-a-new-parameter.html), i.e. for a replacement of https://github.com/ArduPilot/ardupilot_wiki/blob/master/dev/source/docs/code-overview-adding-a-new-parameter.rst.

It describes more completely the addition of parameters to a newly created library (at least as much as it worked for me LOL, please critically review). The chapter on existing libraries was changed accordingly.

I would think that this topic would be better placed in the general ArduPilot codebase, I assume it's not just valid for Copter.

Sorry that this is no PR, I have no idea how to do that, maybe it's useful nevertheless ;).

For your convenience here is a link to download the complete file:
www.olliw.eu/drop/code-overview-adding-a-new-parameter---modifiedbyolliw.rst

Cheers, Olli

-------------------------------------------

.. _code-overview-adding-a-new-parameter:

======================================
Adding a New Parameter (Code Overview)
======================================

Parameters can either be part of the main code or part of a library.

Adding a parameter to the main code
===================================

**Step #1:** Find a spare slot in the Parameters class's enum in
`Parameters.h <https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/Parameters.h>`__
and add your new parameter as shown below in red.  Some things to be
careful of:

-  Try to add the parameter close to parameters that share a similar
   function or worst case add to the end of the "Misc" section.
-  Ensure that the section you are adding it to is not full.  You can
   check if the section is full by counting the number of parameters in
   the section and ensuring that the last element is less than the next
   sections first element.  I.e. the Misc section's first parameter is
   #20.  my_new_parameter is #36.  If the next section began at #36 we
   would not be able to add the new parameter here.
-  Do not add a parameter in the middle of a group thus causing the
   slots for existing parameters to change
-  Do not use slots with "deprecated" or "remove" comments beside them
   because some users may still have the defaults for these old
   parameters in their eeprom so your new parameter's default value
   could be set strangely.

::

        enum {
            // Misc
            //
            k_param_log_bitmask = 20,
            k_param_log_last_filenumber,            // *** Deprecated - remove
                                                    // with next eeprom number
                                                    // change
            k_param_toy_yaw_rate,                           // THOR The memory
                                                            // location for the
                                                            // Yaw Rate 1 = fast,
                                                            // 2 = med, 3 = slow

            k_param_crosstrack_min_distance,    // deprecated - remove with next eeprom number change
            k_param_rssi_pin,
            k_param_throttle_accel_enabled,     // deprecated - remove
            k_param_wp_yaw_behavior,
            k_param_acro_trainer,
            k_param_pilot_velocity_z_max,
            k_param_circle_rate,
            k_param_sonar_gain,
            k_param_ch8_option,
            k_param_arming_check_enabled,
[... 295 lines cut ...]
        // @Values: 0:None,1:coolnewfunction
        // @User: Standard
        AP_GROUPINFO("_PARAM1", 0, AP_MyOwnLib, _my_new_parameter1, MY_NEW_PARAM1_DEFAULT),

        // @Param: _PARAM2
        // @DisplayName: 
        // @Description: 
        // @Range: -1 1
        // @Increment: 1
        // @User: Advanced
        AP_GROUPINFO("_PARAM2", 1, AP_MyOwnLib, _my_new_parameter2, MY_NEW_PARAM2_DEFAULT),

        AP_GROUPEND
    };
    
    /// Constructor
    AP_MyOwnLib::AP_MyOwnLib()
    {
        AP_Param::setup_object_defaults(this, var_info);

        <snip>
    }

This concludes the procedure. The parameter can now be accessed from within the library or
the main code like any other class variable.

hamishwillee, 2016-08-01:
Hi @olliw42 

Thanks very much for taking the time to post this. First thing, this is actually on the DEV wiki - the real URL is here: http://ardupilot.org/dev/docs/code-overview-adding-a-new-parameter.html
The reason you're seeing it on copter is that things get redirected in odd ways from the old ardupilot.com site to the new ardupilot.org site.

The instructions for making [edits/PRs are here](http://ardupilot.org/dev/docs/common-wiki_editing_guide.html) ... this appears on all wikis from the top menu "about" link. Would it be possible to "remake" your changes following these instructions? (you can probably just copy-paste what you have now into an edit window). That will make it much easier for me to review the differences.

Regards
Hamish

olliw42, 2016-08-02:
ah ... I didn't realize that there are extra info for the wiki ... this is very well made, indeed very easy !! (when I read the info for PR to the code I passed out, but this could do LOL)

closed 2016-08-03 by olliw42