ArduPilot/ardupilot_wiki #509: MAVLink Commands - Broken Link

Fossdal, 2016-08-03:
[MAVLink Commands](http://ardupilot.org/dev/docs/mavlink-commands.html) 

Broken link for "MavLink Tutorial for Absolute Dummies (Part–1) (from DIY Drones)." 

when i click i get 

Not Found

The requested URL /downloads/wiki/pdf_guides/MAVLINK_FOR_DUMMIESPart1_v.1.1.pdf was not found on this server.

Apache/2.4.7 (Ubuntu) Server at firmware.ardupilot.org Port 80

hamishwillee, 2016-08-07:
@jpkh Can you please make this available with our other hosted files (I found it here: http://api.ning.com/files/i*tFWQTF2R*7Mmw7hksAU-u9IABKNDO9apguOiSOCfvi2znk1tXhur0Bt00jTOldFvob-Sczg3*lDcgChG26QaHZpzEcISM5/MAVLINK_FOR_DUMMIESPart1_v.1.1.pdf). If not, I can simply include in the repo.

peterbarker, 2016-11-15:
Fixed the link to reference ning.  Its a start.

rmackay9, 2017-01-01:
I'm going to close this because the link seems to work now.

closed 2017-01-01 by rmackay9