ArduPilot/ardupilot_wiki #516: Setting up SITL on Linux

jhljx, 2016-08-15:
[Setting up SITL on Linux](http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

I couldn't install SITL successfully on my Linux. the directory of ardupilot/ArduPlane doesn't has file sim_vehicle.sh. And the instruction sim_vehicle.sh -w couldn't run normally. Could the documentation be more detailed???

hamishwillee, 2016-08-15:
Yes, the instructions need an update - I just found out today that sim_vehicle.sh has been replaced by sim_vehicle.py. 
The file is not (and never was) in the plane directory. Instead it is in /test/autotest which you add to your system paths (that's what these lines in the .bashrc instructions are about):
```
export PATH=$PATH:$HOME/jsbsim/src
export PATH=$PATH:$HOME/ardupilot/Tools/autotest
export PATH=/usr/lib/ccache:$PATH
```
@peterbarker Would you be able to scan the instructions linked and confirm that the only probable change is to the file extension of sim_vehicle. I hope to test this, but having someone who has tried it first confirm would make life easier.

jhljx, 2016-08-15:
Thank you very much,I have been troubled by this a whole day. But I still don't know how to install the SITL. And when I type 'python sim_vehicle.py' or 'python sim_vehicle.py -w' in the terminal, it has some error as follows:

SIM_VEHICLE: Start
SIM_VEHICLE: ** Is (autotest) really your vehicle type?  Try  -v VEHICLETYPE  if not, or be in the e.g. ArduCopter subdirectory
Traceback (most recent call last):
  File "sim_vehicle.py", line 329, in <module>
    cmd_opts.frame = default_frame_for_vehicle[cmd_opts.vehicle]
KeyError: 'autotest'
SIM_VEHICLE: Killing tasks

I feel it so hard to install it.

hamishwillee, 2016-08-15:
What directory are you calling `python sim_vehicle.py -w` in? If you have set up your paths then you should call it from the directory of the vehicle you want to build (e.g. /ardupilot/copter).
If you are calling this from the autotest directory then you need to specify your vehicle as the error says
- so in this case something like:
```
python sim_vehicle.py -w -v copter
```
(not sure about the valid types, but it will be one of Copter, copter, ArduCopter.)

jhljx, 2016-08-15:
I run the instruction 'python3 sim_vehicle.py -w -v ArduCopter',and it show that ' FileNotFoudError:No such file or directory: /ardupilot/modules/waf/waf-light',then I 'git clone https://github.com/ArduPilot/waf.git' in the directory 'waf',it runs smoothly.And the terminal show that

GPS lock at 0 meters
APM: EKF2 IMU1 Origin Set
APM: EKF2 IMU0 Origin Set
APM: EKF2 IMU0 is using GPS
APM: EKF2 IMU1 is using GPS
Flight battery 100 percent

then I type a 'Enter' on my keyboard.It shows 

'STABILIZE> '

And I don't know how to do next ??

magicrub, 2016-08-15:
Sounds like you got it working. Next step is to follow the wiki on how to use SITL. Asking questions on how to use it should be done at discuss.ardupilot.org instead of this bug tracker system. Closing.

closed 2016-08-15 by magicrub

jhljx, 2016-08-15:
Thanks a lot