ArduPilot/ardupilot_wiki #517: Setting up SITL on Linux

jhljx, 2016-08-15:
Setting up SITL on Linux - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

And the instruction './autogen.sh --enable-libraries'  couldn't run successfully

magicrub, 2016-08-15:
Can you provide more information? Perhaps paste here the output of running the command?

jhljx, 2016-08-15:
After I type the instruction './autogen.sh --enable-libraries' It shows as follows:

processing .
deletefiles is 
Running libtoolize...
Remember to add `AM_PROG_LIBTOOL' to `configure.in'.
You should add the contents of `/usr/local/share/aclocal/libtool.m4' to `aclocal.m4'.
Running aclocal -I .  ...
aclocal: warning: autoconf input should be named 'configure.ac', not 'configure.in'
aclocal: error: couldn't open directory 'm4': No such file or directory

**Error**: aclocal failed. This may mean that you have not
installed all of the packages you need, or you may need to
set ACLOCAL_FLAGS to include "-I $prefix/share/aclocal"
for the prefix where you installed the packages whose
macros were not found

magicrub, 2016-08-15:
Why are you calling auto gen? For sitl you run sim_vehicle.py from writing
the vehicle folder such as /ArduPlane.

Closing because this is another help question in the wiki bug tracker
system.

closed 2016-08-15 by magicrub