ArduPilot/ardupilot_wiki #520: http://ardupilot.org/plane/docs/automatic-flaps.html#using-flaps

CraigElder, 2016-08-22:
To use flaps they must be configured on channel 5, 6, 7 or 8. If you wish to have manual control of the flaps then they must be on 5, 6, or 7.

Is out of date.  Flaps can be on any channel

hamishwillee, 2016-08-23:
Hi @CraigElder 
So should the line just change to:

> Flaps can be used on any channel.

Ie there is no channel for configuration, and no special subset for manual control? 

Is there a version that this changed in? AP 3.? say

CraigElder, 2016-08-26:
>>>Ie there is no channel for configuration, and no special subset for manual control?
That is correct
>>>Is there a version that this changed in? AP 3.? say
I don't remember. This changed nearly a year ago

I would also like to suggest that we add a recommended procedure for testing the auto flap function.

Remove the prop
Arm the vehicle
Select Loiter Mode
Set the airspeed to a speed to a value below the FLAP_1_SPEED and the flaps should deploy.

hamishwillee, 2016-08-26:
I did the quick fix for initial problem [here](https://github.com/ArduPilot/ardupilot_wiki/commit/e2c161e188f876b04573703c17802cfcf0c6577b?w=1). 

Can you add the testing section as a PR?

brunoolivieri, 2019-08-20:
There was some time since this issue opening and some updates.

Is this issue still actual and need to be followed, or we could close it?

Hwurzburg, 2019-10-10:
flaps/flaperons recently updated....closing

closed 2019-10-10 by Hwurzburg