ArduPilot/ardupilot_wiki #521: Cygwin Install  "C:\cygwin\home" folder empty

jhenry5, 2016-08-22:
Hi, I followed the instructions http://ardupilot.org/dev/docs/sitl-native-on-windows.html  to install 32 bit cygwin on my 64 bit windows 8.1 laptop. My next step is to open my .bashrc file located in C:\cygwin\home\ & add a line to it. The problem I'm having is that " C:\cygwin\home\" is empty when i open it in windows explorer.
1) I unhid all hidden files. 
2) I re-installed cygwin for a total of three installs from different servers.
I cant find .bashrc file to modify, its not in C:\cygwin\home. **This folder is empty.**Please help

hamishwillee, 2016-08-22:
Hi @jhenry5 

Firstly, best place to get support for development is the [discussion boards](http://discuss.ardupilot.org/). That said, this is probably a docs issue too.

So the file should actually be C:/cygwin/home/**user_name**/.bashrc , but obviously that doesn't help you since your home folder is empty. Perhaps the best thing to do is work out where cygwin things your home folder is. Please open cygwin terminal and type the following:
```
cd ~
pwd
ls -la
```
The first line takes you to your home directory, the second line tells you where that is and the third lists all files. If it includes a .bashrc file then you know it exists and can search for the folder name in your normal drive (or edit via the prompt).

If the .bashrc is not there, then we have a problem. Let me know what you see.

jhenry5, 2016-08-23:
Thanks kindly,
The home folder isnt where i expect it to be. But the files, i require arent in this folder either.

[image]

Looking at the file creation dates bash.history is the only file modified today. (8/22/2016)

Any feedback will be greatly appreciated

hamishwillee, 2016-08-23:
Hi @jhenry5 

That is very odd - it might be worth uninstalling/reinstalling Cygwin. 

When you first opened the cygwin terminal (step 1 in "Set up directories/paths in Cygwin") it should have showed you where it was creating the .bashrc file. If it didn't this step may have failed. I'd be searching for the file across my C drive.

Other than that I see two options:

1. Just create a file .bashrc in that directory and add the information. This may work, but I'm concerned that there may be other useful information in the default .bashrc that you are missing.
2. Start using Google search to find out how to change the home folder.  E.g. [see this](http://stackoverflow.com/questions/1494658/how-can-i-change-my-cygwin-home-folder-after-installation). You would probably do this anyway since a dia folder is not an ideal home location.

I think that's pretty much the help I can give you - Cygwin is just part of the toolchain - not really an "ardupilot" problem.

jhenry5, 2016-08-23:
cygwin doesnt have an uninstall option and does not show up in my add remove programs.
I deleted the "C/Cygwin" folder & tried again, without success.....My install selections are below:
[image]
[image]

[image]

[image]

**The full print out of my packages**
autoconf2.1	(2.13-12)
	Stable version of the automatic configure script builder
	Required by: autoconf, automake1.4, automake1.5

autoconf2.5	(2.69-3)
	Current version of the automatic configure script builder
	Required by: autoconf, automake1.10, automake1.11, automake1.12, automake1.13, automake1.14, automake1.15, automake1.6, automake1.7, automake1.8, automake1.9

automake1.10	(1.10.3-2)
	(1.10) a tool for generating GNU-compliant Makefiles
	Required by: automake

automake1.11	(1.11.6-2)
	(1.11) a tool for generating GNU-compliant Makefiles
	Required by: automake

automake1.12	(1.12.6-2)
	(1.12) a tool for generating GNU-compliant Makefiles
	Required by: automake

automake1.13	(1.13.4-1)
	(1.13) a tool for generating GNU-compliant Makefiles
	Required by: automake

automake1.14	(1.14.1-2)
	(1.14) a tool for generating GNU-compliant Makefiles
	Required by: automake

automake1.15	(1.15-1)
	(1.15) a tool for generating GNU-compliant Makefiles
	Required by: automake

automake1.4	(1.4p6-11)
	(1.4) a tool for generating GNU-compliant Makefiles
	Required by: automake

automake1.5	(1.5-11)
	(1.5) a tool for generating GNU-compliant Makefiles
	Required by: automake

automake1.6	(1.6.3-12)
	(1.6) a tool for generating GNU-compliant Makefiles
	Required by: automake

automake1.7	(1.7.9-11)
	(1.7) a tool for generating GNU-compliant Makefiles
	Required by: automake

automake1.8	(1.8.5-11)
	(1.8) a tool for generating GNU-compliant Makefiles
	Required by: automake

automake1.9	(1.9.6-11)
	(1.9) a tool for generating GNU-compliant Makefiles
	Required by: automake

binutils	(2.25-4)
	GNU assembler, linker, and similar utilities
	Required by: gcc-core, python

[... 314 lines cut ...]
	Required by: automake1.10, automake1.11, automake1.12, automake1.13, automake1.14, automake1.15, automake1.4, automake1.5, automake1.6, automake1.7, automake1.8, automake1.9

w32api-headers	(4.0.4-1)
	MinGW-w64 Windows API headers for Cygwin
	Required by: gcc-core, w32api-runtime

w32api-runtime	(4.0.4-1)
	MinGW-w64 Windows API import libraries for Cygwin
	Required by: gcc-core

windows-default-manifest	(6.4-1)
	Default Windows application manifest
	Required by: gcc-core

xz	(5.2.2-1)
	LZMA de/compressor
	Required by: tar, automake1.10, automake1.11, automake1.12, automake1.13, automake1.14, automake1.15

zlib-devel	(1.2.8-3)
	gzip de/compression library (development)
	Required by: binutils

zlib0	(1.2.8-3)
	gzip de/compression library (runtime)
	Required by: ccache, file, gcc-g++, git, libopenssl100, man-db, util-linux, gcc-core, libcurl4, openssh, python, binutils, libssh2_1, zlib-devel

hamishwillee, 2016-08-23:
Try installing it for "just you". I'll bet that might set the default to your home directory.

jhenry5, 2016-08-23:
I reinstalled cygwin 4 times and C:\cygwin\home is still empty.  I then tried it again with the "just for you" setting, unsuccessfully. How do I generate all the files(including .bashrc) for C:\cygwin\home folder. 1)Can you point me(& anyone else with this problem) in the right direction to generate this file?﻿ 2) Where can you find the cigwyn uninstall "button" ?

Thx alot mate, I've been at this from 8 am to midnight

hamishwillee, 2016-08-23:
Hi @jhenry5 

Sorry you're having so much trouble. Google is your friend!!!

I found [this post](http://superuser.com/questions/584540/no-bashrc-file-in-my-home-directory) which suggests that the default files are found in `\etc\skel`. Doing a search I find them in my drive at `C:\cygwin\etc\skel`. I also find that there is no "button" for uninstalling Cygwin - but there are some instructions here: https://cygwin.com/faq/faq.html#faq.setup.uninstall-all

So you first need to set your home and then move these files in there.

Good luck

Regards
H

peterbarker, 2016-11-15:
@jhenry5 Were your problems eventually resolved?

Are there any modifications we need to make the the ArduPilot Wiki so this doesn't happen to someone else?

kshitijvyas1985, 2018-01-05:
Any body facing this issue. Just set there environment variable
Right click on "My Computer" -> Properties -> Advanced -> Environment Variables
Add a new environment variable, called CYGWIN_HOME and set its value to C:\cygwin
Edit the PATH environment variable and add C:\cygwin\bin to it (usually separated by a ';').
Just click okay, exit any command prompts or bash shells (over cygwin) you may have open, and open it again - it'll work!
Assumption - this assumes that you have installed cygwin at C:\cygwin. If you've kept it someplace else, please modify the above accordingly

brunoolivieri, 2019-08-23:
@jhenry5 Were your problems eventually resolved?

@kshitijvyas1985 I did the PR #1930 in order to emphasize the path configuration.

Moreover, this page has been updated since the issue opening. Makes sense remain with it open?

Hwurzburg, 2020-01-15:
this seems resolved...closing

closed 2020-01-15 by Hwurzburg

ieliopou0, 2022-04-12:
Just adding this comment in case anyone else has this issue.
I also had this issue, where  C:\cygwin64\home was empty. The terminal displayed "-bash-4.4$" as the prompt, and some basic commands wouldn't work like "ls" and "clear". Adding the cygwin64 bin folder to the PATH environment variable like kshitijvyas1985 mentioned helped get some commands like "ls" to work, but home folder was till empty, "clear" command still didn't work, and the terminal prompt was still weird. Looked up how to at least get the "clear" command to work, and someone mentioned install the "ncurses" packages. So I re-ran the installer, kept the same folders I used to install cygwin originally, and then looked for packages that were not installed and searched for "ncurses", then installed the latest version of all the packages listed. I also noticed just before looking for ncurses stuff that there were a lot of packages listed under "pending". Either way I hit install, and this time when I opened up my cygwin terminal it looked normal, and the home directory had the .bashrc and other expected files under home/<user>. I'm new to cygwin so I don't know if those pending packages meant they weren't installed yet, and I had to re-run the installer to install them, or if it was due to the ncurses stuff, but it worked for me. Hopefully it works for anyone else with this issue.