ArduPilot/ardupilot_wiki #531: Table of Contents

CraigElder, 2016-08-30:
[Table of Contents](http://ardupilot.org/plane/docs/common-table-of-contents.html) - 
TOC is broken all of the subsections and all of the parameters are showing up in the TOC

hamishwillee, 2016-08-30:
The full table of contents is only properly generated if you put it on the top/home page - like this one: http://ardupilot.org/planner/index.html

If you generate it on another page using the various options you get a flat list. This is something to do with sphinx that I have not tried to debug. I am sure it is a sphinx issue and not a "use of the tool" issue.

One option is to remove the page (arguably we don't need it). Alternatively add the full index in the top page.

gmorph, 2016-11-15:
I'd just close this issue and assume its a limitation of the tool.  Probably not worth our time to investigate and fix.

hamishwillee, 2016-11-16:
Well, one option is to remove the TOC page - we don't really need it when you have the sidebar. I just didn't want to do that unilaterally.

gmorph, 2016-11-17:
I'm not fussed either way.  Does anyone ever load the TOC and then search it via the browser for a keyword?  Anyway as I said remove the TOC page or don't - I'm fine either way.
Thanks, Grant.

Hwurzburg, 2019-10-10:
I think TOCs are okay now on all the sites....closing

closed 2019-10-10 by Hwurzburg