ArduPilot/ardupilot_wiki #532: how to add "edit on github"on the website

riddle911, 2016-08-30:
@hamishwillee 
Hi Willee,
I'm working with HEX TECH and many other developers to translate the offical wiki to Chinese.
Now I have built the wiki using syphinx as yours(wiki.osdrone.net), but I still come across two problems. Hope you can give me the solution.
1.How did you add the menu on the top of the website.
2.How to add the button "edit on github" on the web.
All my websites are made by "$ make html" .  I 'm very confused about this  and it's hard to find anyone to ask something on Syphinxdoc.  I have also digged in syphinx's documentation but comes no result.
Hope for your reply. Thank you!

hamishwillee, 2016-08-30:
Hi @riddle911 
The answer to both questions is that I forked the sphinx rtd theme - you can see my version here: https://github.com/ArduPilot/sphinx_rtd_theme

The readme in that site explains how I did it (reproduced below) and you can examine the source:

----

This is a modification of the [snide/sphinx_rtd_theme](https://github.com/snide/sphinx_rtd_theme). It supports the addition of a top site bar above the main sphinx content. This site bar includes menu content that is populated based on the presence of certain files in the source tree at /docs/. For example, if a file docs/project-news.rst is present in the tree an "About" menu will be displayed.

It also will display a table of contents in any article named /docs/common-table-of-contents.rst

The theme displays the "Edit on Github" link by default on non-index pages. This is VERY tied into the ardupilot_wiki source structure. It uses the value in "html_short_title" field in the conf.py to determine the path. The value must therefore be a path that matches where the source is on github - e.g. "copter", "rover", "ardupilot" If this is done, source files will correctly be linked in https://github.com/hamishwillee/ardupilot_wiki. For common files it will link to the original in the common folder.

----
> All my websites are made by "$ make html" . I 'm very confused about this and it's hard to find anyone to ask something on Syphinxdoc. I have also digged in syphinx's documentation but comes no result.

OK so this is a little more complicated to explain. Sphinx builds the content in the current conf.py directory when you do "make html". However our wiki is more complicated in that 

* we actually have to build a number sites (copter, rover, etc)
* some of the content is "common" and appears in a number of sites
* Some of the content comes from parameters in source, so we have to download it from a server.

So in the root of ardupilot_wiki you'll see a file called update.py. This can be called inside a linux environment to copy the common pages to each site folder (cleaning out any existing first), download the parameter files and then build all of the sites. This is all run automatically on the site. Hope that helps.

There are other poeple translating the content too - do you known 白彧实 - "Stone Bai"? It may be that you guys are doing the same thing separately.

riddle911, 2016-08-30:
@hamishwillee 
WOW~I'm now very very excited! I made it! Thank you Willee!

I just update your version sphinx_rtd_theme and finally it works! 
Now our team could do the translation more conveniently.

I know Stone Bai on the Internet and we also have  some communication. He with other volunteers have translated the old apm wiki and some of the ardupilot wiki.
 
But for some reason, he just stopped or he didn't carry on with the translation.

My team just translated some papers and the dev.px4.io for developers.

So together with more professional developers, we decide to make up the translation of Ardupilot wiki.

And we also want to build a dev community for Chinese drone developers under Hextech's support.

I believe this community will become much more stronger and many of Chinese developers are eager to communicate or even to talk with you guys. lol.

Thank you again! And  I hope one day you could talk with Chinese fans.

hamishwillee, 2016-08-30:
@riddle911 Excellent! Stone Bai did find some issues with translation. He was using Transifex for the centralised translation server and found that often the way that sphinx does line breaks was not helpful for translation. 

You might want to look at the standard Sphinx translation mechanisms since this would allow you to make translation files and use "virtually" the same build process and me (and remain in sync).

I would love to talk to Chinese fans, but the best people would be the core developers for each platform. Are you in touch with Craig Elder (our community manager). He might be able to help organise something in future! Email me at hamishwillee at gmail dot com if you need the contact!

gmorph, 2016-11-15:
This looks resolved - can we close it out?

closed 2016-11-16 by riddle911