ArduPilot/ardupilot_wiki #533: Ubuntu - sudo apt-get install python & pymavlink directory

jhenry5, 2016-08-31:
Hi,

I am installing SITL on a Ubuntu 14.04 running on VMware running on windows 8,1, 64 bit as shown here: http://planner.ardupilot.com/dev/docs/setting-up-sitl-on-linux.html#setting-up-sitl-on-linux

I am at this stage :
**Install some required packages**

If you are on a debian based system (such as Ubuntu or Mint) then run this:

sudo apt-get install python-matplotlib python-serial python-wxgtk2.8 python-lxml
sudo apt-get install python-scipy python-opencv ccache gawk git python-pip python-pexpect
sudo pip install pymavlink MAVProxy

What directory do I call these commands above in ?  home , ardupliot, jsbsim ?
The instructions don't specify.

magicrub, 2016-08-31:
shouldn't matter, have you tried it yet? Also, there's `ardupilot\Tools\scripts\install-prereqs-ubuntu.sh ` which I think does all that for you. Is that in the wiki? If not we should add it.

billbonney, 2016-08-31:
I think you might find that you need a newer gcc that is supplied with 14.04.  16.04 might be a better choice. 

The whole wiki page needs updating. We should have these support request on http:://discuss.ardupilot.con

magicrub, 2016-08-31:
that script should do a gcc/g++ check and upgrade or throw an error

hamishwillee, 2016-08-31:
@billbonney 

> We should have these support request on http:://discuss.ardupilot.com

Absolutely. However the fact that it came here does highlight the need to update the page. 

So based on this I presume the requirement is to start with Ubuntu 16.04. Also does the script `ardupilot\Tools\scripts\install-prereqs-ubuntu.sh` do all the setup stuff described in the current doc (ie Jbsim, exporting paths, etc?) and is it maintained? Yes, I will check but just wanted "basically yes".

Cheers

billbonney, 2016-09-01:
no idea, i usually just install the pre-reqs by hand `sudo apt instal XXX`

jhenry5, 2016-09-01:
Thanks for the feedback. 
I downloaded an ubuntu 14.04 image as this was recommended  in the link I posted earlier.
http://planner.ardupilot.com/dev/docs/setting-up-sitl-on-linux.html#setting-up-sitl-on-linux

While the link I posted with step by step instructions is being updated, is there another website with step by step instructions on installing SITL ? Can someone share it?
Essentially I am trying to visualize DroneKit commands in the Gazebo simulator.  

I am running ubuntu  14.04 on windows 8.1
As far as I am aware, these are the instructions I need to follow:

**SITL on linux** : http://planner.ardupilot.com/dev/docs/setting-up-sitl-on-linux.html
**SITL Gazebo**: http://ardupilot.org/dev/docs/using-rosgazebo-simulator-with-sitl.html
**SITL Dronekit** : http://python.dronekit.io/develop/sitl_setup.html

But apparently I'm using outdated links.  Links with  relevant updated step by step instructions will be greatly appreciated.

Sorry, i'm completely new to this.
Thanks alot guys

magicrub, 2016-09-01:
It's not that the links are outdated, it's just that development goes faster than the documentation does. Please let us know any troubles you have and we'll investigate. We don't have time to go through all the tutorials every time we change the code. When it breaks we fix it. Let us know if/where it's broken and we'll jump on it.

hamishwillee, 2016-09-01:
@jhenry5 PS those links don't "all" need to be followed. You only need the second topic if you want to use ROS/Gazebo. The first and last topics both allow you to launch SITL, but the first does this from source on Linux while the Dronekit version launches pre-built SITL binaries on Windows, Linux, Mac OS on X86 architectures (ARM binaries not supported).

The last is by far the easiest if all you want to do is have a version of SITL to test dronekit code against, and you don't care if the simulation is for the bleeding edge of ardupilot.

davidbuzz, 2016-11-15:
this issue seems to be mostly an installation issue for a single user at this point.    resolving this should really happen at http://discuss.ardupilot.org  not in the wiki issues, so I'm closing this issue for now.

closed 2016-11-15 by davidbuzz