ArduPilot/ardupilot_wiki #536: Choosing a Ground Station

auturgy, 2016-09-03:
[Choosing a Ground Station](http://ardupilot.org/copter/docs/common-choosing-a-ground-station.html) - Fightingwalrus.com is a dead link

hamishwillee, 2016-09-04:
Thank you. True. 

@rmackay9 My temptation is to remove this entry. Their site is down (all the time). The app hasn't had an update for a year. Thoughts?

CraigElder, 2016-09-06:
I suggest we can remove it. Fighting Walrus is gone.

rmackay9, 2016-09-07:
fine with me.

hamishwillee, 2016-09-11:
Thank you. This has been removed.

closed 2016-09-11 by hamishwillee

auturgy, 2016-09-11:
No worries, thanks Hamish.
There are a number of other things spread throughout the wiki's that have been overtaken by time. I'm more than happy to help out reviewing and updating if you could let me know the process to do so.
Regards,

James

hamishwillee, 2016-09-12:
Hi James

That would be excellent! The process is pretty simple make a PR and we'll get reviewed. The PR should explain your reasoning in the same way this issue did.

The editing guide explains most of what you might need to know: http://ardupilot.org/copter/docs/common-wiki_editing_guide.html

IF you get stuck, you can always ask for help here or on the discussion boards.

skinnyb8, 2017-01-20:
I Really Appreciate your responses to me… it’s the first offer of help I’ve received in four years of trying to accomplish this…

 

First thing that happened is the APM I ordered was a much older configuration and no instructions came with it… I promptly smoked the first one but re ordered and spent a year trying to get it running.

But Now I can’t use it and have over $300 tied up in it… haven’t touched it since last year.

 

It said I could us version 3.0 and it did work perfectly for a little over an hour before the Mission Planner Software automatically updated…  

 

I sent email to several of the You tube Subscriptions that I watched to finally get it running… but nobody can tell me where to get the older version of Mission Planner and turn of Automatic update.

 

I’m working with Arduino on other small electronic “Projects”… but have all but given up on the APM 2.8?? (I think that’s right)

 

Again, I really appreciate being able to “talk” to someone about this…

 

Kenny B

hamishwillee, 2017-01-30:
@skinnyb8 Please don't ask questions on github that piggy back on completely unrelated discussions. The right place for this sort of thing is http://discuss.ardupilot.org/