ArduPilot/ardupilot_wiki #538: Building ArduPilot for Pixhawk/PX4 on Windows with Make

Metello, 2016-09-04:
[Building ArduPilot for Pixhawk/PX4 on Windows with Make](http://ardupilot.org/dev/docs/building-px4-with-make.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

When I built the code  ,I got this :
 fatal: Not a git repository: f:/MyRepository/3.3/ardupilot/modules/../../.git/modules/modules/PX4Firmware
Failed to get px4 hash
fatal: Not a git repository: f:/MyRepository/3.3/ardupilot/modules/PX4NuttX/nuttx//../../.git/modules/modules/PX4NuttX
Failed to get nuttx hash
PX4 ArduCopter Firmware is in ArduCopter-v2.px4

Does it matter？Did I need resovle this? How?  
Thank you very much!!

magicrub, 2016-09-04:
That is normal and we should note that in the wiki that as long as you get "Firmware is.." then a final firmware was created and you're good to go.

closed 2016-09-04 by magicrub

magicrub, 2016-09-04:
@Metello I've updated the wiki. Thanks for the comment!

Metello, 2016-09-05:
Thankyou very much!!

louise1983, 2018-02-12:
Hello ... i have gotten "git clone ardupilot"by git bash , and still trying to built it by eclipse and px4toolchain ....i got some error:
make[1]: *** [/c/git1/ardupilot/modules/PX4Firmware/Archives/px4fmu-v2.export] Error 7
make: *** [/c/git1/ardupilot/modules/PX4Firmware/Archives/px4fmu-v2.export] Error 2
make[1]: Leaving directory `/c/git1/ardupilot/modules/PX4Firmware'

anyone help me?