ArduPilot/ardupilot_wiki #541: Add link to donate and partners pages below donate button

rmackay9, 2016-09-05:
[image]

jpkh, 2016-09-05:
Ok that will be added on next template set update.

hamishwillee, 2016-09-05:
One point to make with adding this here is that the TOC for the page itself will not be displayed in place. That is because it isn't part of the actual document TOC.

The same thing happens if articles are in the top menu but not the document TOC (which is why I usually add to both) but you can't reasonably do that here because then you'd have the documents appearing twice in the sidebar. 

Not suggesting you have to change this, but an alternative would be to have these links in the menu "Donate" and then include them in the appendix so that they appear properly in the sidebar.

rmackay9, 2016-09-06:
@hamishwillee, how do we add the document to the TOC?  I had a short look around for a document named "commong-toc" or similar but didn't immediately find it.
Does it have something to do with the little tag at the top of the pages?  For example, this appears at the top of autotune.rst, ".. _autotune:"

hamishwillee, 2016-09-06:
Pages are added to the TOC by listing them in the toctree directive at the bottom of the parent page. In this case you'd add it to the common-Appendix. Check out the existing structure of the toctree - if the pages won't appear on all sites you'll have to add some [site] guards specifying which wikis you want the doc to appear in the location.

Note that if you decide to add the docs into the top menu, this requires tweeking of the theme. I basically have code in there to display menu items IFF a particular page exists.

Enough?

jpkh, 2016-09-06:
links have been added to sidebar.

closed 2016-09-06 by jpkh

CraigElder, 2016-09-06:
@hamishwillee Do you think we need more detail about how to work with the TOC here: http://ardupilot.org/ardupilot/docs/common-wiki_editing_guide.html

hamishwillee, 2016-09-06:
@CraigElder Yes, definitely. Created a reminder here: https://github.com/ArduPilot/ardupilot_wiki/issues/544