ArduPilot/ardupilot_wiki #543: Rewrite "SITL on Windows native" topic to use new Windows 10 bash

hamishwillee, 2016-09-05:
As it says in the heading. From lucas

- you have to install xming in order to see the windows
- it should also work to build for pixhawk

davidbuzz, 2016-11-15:
The SITL/windows page in question is: https://github.com/ArduPilot/ardupilot_wiki/blob/master/dev/source/docs/sitl-native-on-windows.rst  and is specifically written for windows 8 user/s.
The windows10/Bash build environment is still experimental, but is found here:
https://github.com/ArduPilot/ardupilot_wiki/blob/master/dev/source/docs/building-ardupilot-onwindows10.rst

I think the best thing to do here is leave the first/old page as-is, and highlight it's for Windows 8 ( and earlier), while creating a new "SITL on windows 10" page for it, as the process will be very different.( and is as yet not documented or even tested really. )

auturgy, 2017-10-15:
Suggest closing this, as Bash in windows requires "workarounds" (aka hacks) to get graphical output, and the existing SITL via cygwin, and SITL with MissionPlanner, both function fine - so I don't see it as an issue per se. Of course if someone wants to contribute a guide, it would be welcome.

closed 2017-10-15 by davidbuzz