ArduPilot/ardupilot_wiki #546: Linux SITL error msg:  no such option --aircraft

jhenry5, 2016-09-08:
I used http://ardupilot.org/dev/docs/setting-up-sitl-on-linux.html instruction to install SITL on ubuntu 14.04.

I ran    **

> sim_vehicle.py -w

** which came to stop, then I pressed Ctrl C as instructed terminating the process.

Next I ran  **

> sim_vehicle.py --console --map --aircraft test

**    where I got the error msg:

**> no such option: --aircraft
> Perhaps you want --sim_vehicle_sh_compatible (-C)?
> SIM_VEHICLE: Killing tasks
> SIM_VEHICLE: kill_tasks failed: global name '_options_for_frame' is not defined**

 .      A full terminal printout below

jh@jh-Lenovo-Ideapad-Z50-70:~/ardupilot/ArduPlane$ sim_vehicle.py -w
SIM_VEHICLE: Start
SIM_VEHICLE: Get JSBSim version
SIM_VEHICLE: "JSBSim" "--version"
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at -35.363261,149.165230,584,353 (CMAC)
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/jh/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Setting top to                           : /home/jh/ardupilot 
Setting out to                           : /home/jh/ardupilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : sitl 
Checking for 'g++' (C++ compiler)        : /usr/lib/ccache/g++ 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/gcc 
Checking for need to link with librt     : not necessary 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : no 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : no 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : yes 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version >= 2.7.0           : 2.7.6 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Checking for program 'size'                    : /usr/bin/size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
'configure' finished successfully (0.522s)
SIM_VEHICLE: Building
SIM_VEHICLE: "/home/jh/ardupilot/modules/waf/waf-light" "build" "--target" "bin/arduplane"
Waf: Entering directory `/home/jh/ardupilot/build/sitl'
Waf: Leaving directory `/home/jh/ardupilot/build/sitl'

BUILD SUMMARY
Build directory: /home/jh/ardupilot/build/sitl
Target         Text     Data  BSS    Total  
--------------------------------------------
bin/arduplane  1250207  1964  36096  1288267

Build commands will be stored in build/sitl/compile_commands.json
'build' finished successfully (1.341s)
SIM_VEHICLE: Using defaults from (/home/jh/ardupilot/Tools/autotest/default_params/plane-jsbsim.parm)
SIM_VEHICLE: Run ArduPlane
SIM_VEHICLE: "/home/jh/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduPlane" "/home/jh/ardupilot/build/sitl/bin/arduplane" "-S" "-I0" "--home" "-35.363261,149.165230,584,353" "-w" "--model" "jsbsim" "--speedup" "1" "--defaults" "/home/jh/ardupilot/Tools/autotest/default_params/plane-jsbsim.parm"
[... 137 lines cut ...]
APM: EKF2 IMU0 Origin Set
APM: EKF2 IMU1 Origin Set
APM: EKF2 IMU0 is using GPS
APM: EKF2 IMU1 is using GPS
Traceback (most recent call last):
  File "/home/jh/ardupilot/Tools/autotest/sim_vehicle.py", line 832, in <module>
    start_mavproxy(cmd_opts, frame_infos)
  File "/home/jh/ardupilot/Tools/autotest/sim_vehicle.py", line 779, in start_mavproxy
    run_cmd_blocking("Run MavProxy", cmd, env=env)
  File "/home/jh/ardupilot/Tools/autotest/sim_vehicle.py", line 630, in run_cmd_blocking
    ret = os.waitpid(p.pid, 0)
KeyboardInterrupt
SIM_VEHICLE: Killing tasks
jh@jh-Lenovo-Ideapad-Z50-70:~/ardupilot/ArduPlane$ sim_vehicle.py --console --map --aircraft test
SIM_VEHICLE: Start
no such option: --aircraft
Perhaps you want --sim_vehicle_sh_compatible (-C)?
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: kill_tasks failed: global name '_options_for_frame' is not defined

 Modified Bashrc file in home directory

[image]

[image]

gmorph, 2016-09-08:
I've replicated this issue.  You are correct that the example shown shows --aircraft as an option but it isn't for sim_vehicle.py but it is a MAVProxy parameter.  I suspect the old sim_vehicle.sh just passed any flags it didn't understand straight onto MAVProxy but the new .py doesn't. If you use the -m flag i.e.
sim_vehicle.py --console --map -m '--aircraft test'
then it works.
I'll assign to Peter for a look.

Thanks, Grant.

jhenry5, 2016-09-08:
Excellent. The program boots up perfectly!
Thanks kindly,

I tried running the first command in the installation guide to ensure the program was installed correctly
[image]

After loading the waypoints and arming the throttle, the system will disarm quickly if nothing else is typed for ~ 2 seconds. 
If I type: "mode auto" in quick succession after "arm throttle". The system disarms instead of flying as the tutorial said it should. 

1) Is this due to improper installation  or simply I'm calling the commands wrong?
2) The **"Learn Mavproxyl"** link embedded in "Linux SITL" link doesnt work ? http://tridge.github.io/MAVProxy/ 

3) Do I simply add DroneKIT and Gazebo files to this ardupilot folder to visualize dronekit commands in gazebo? The gazebo plugin by Alex seems to be a completely
 different ardupilot build, while I currently have different build. 
Alex's version requires that I install : 

[image]
git clone https://github.com/alexbuyval/ardupilot
git checkout RangeFinderSITL2

Thanks so kindly mate

peterbarker, 2016-09-09:
@gmorph is correct in his surmise about the way sim_vehicle.sh worked.

I looked at updating the wiki, but decided to add a "special-case" parameter instead.  These are parameters we recognize and pass through to mavproxy (in addition to those specified with the "-m" option).

So now on master sim_vehicle.py now understands --aircraft and passes that to mavproxy.

Thanks for the bug report @jhenry5 :-)

@jhenry5 You've asked three more questions on here.

1. The disarm thing *should* be 10 seconds.  It is controllable with the "DISARM_DELAY" parameter (set it to zero for no disarm).
2. @tridge may or may not know about that link issue; I'll raise it with him.
3. is beyond my currently knowledge, sorry!  Could you raise a separate issue for that if you can't solve it, please?

closed 2016-09-09 by peterbarker