ArduPilot/ardupilot_wiki #550: Putty Recommendation for Linux on Desktop Pc

jhenry5, 2016-09-15:
The DroneKit tutorial recommends Putty to interface the Raspberry pi and a Desktop pc running windows.
My desktop pc is running ubuntu 14.04. 
What SSH/Telnet Client do you recommend to connect your linux desktop to the raspberry pi ?  ie Where the steps in this http://planner.ardupilot.com/dev/docs/raspberry-pi-via-mavlink.html 
tutorial can be followed?

billbonney, 2016-09-15:
ssh pi@<ipaddresOfPi> from Terminal app. Or a text console. It's much simpler than windows.

earsneyes, 2016-09-15:
You can have putty on ubuntu as well :)

Open a terminal and try the following to install it.

`sudo apt-get install putty`

magicrub, 2016-09-15:
I think this is a true breakthrough for Linux. Someone is running Ubuntu
full-time on their desktop and they don't know about ssh. The Linux
audience is widening. Gooo Linux, it's finally hitting mainstream!

billbonney, 2016-09-15:
i'm just having a lie down... anyway, just install cygwin on windows or the new bash for windows 10 and the world doesn't need PuTTY 👍 

Command Line Rules 😜!!!

jhenry5, 2016-09-15:
_sudo apt get install putty_ worked perfectly in my ubunutu 14( on metal). Thanks, earsneyes.
Less than 3 hrs experience in my dual boot linux, hopefully someone else benefits from my noob questions :)