ArduPilot/ardupilot_wiki #562: Building ArduPilot for Pixhawk/PX4 on Mac with Make

lvale, 2016-10-01:
[Building ArduPilot for Pixhawk/PX4 on Mac with Make](http://ardupilot.org/dev/docs/building-px4-with-make-on-mac.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Outdated

Sparkle:ardupilot Luis$ make px4-clean
make: *** No rule to make target `px4-clean'.  Stop.
Sparkle:ardupilot Luis$ 

also these instructions have problems
https://github.com/ArduPilot/ardupilot/blob/master/BUILD.md

magicrub, 2016-10-01:
I had major problems with this yesterday. Why isn't there a script like we
have for ubuntu?

billbonney, 2016-10-05:
You can build for macOS (OSX) using waf, see http://discuss.ardupilot.org/t/building-ardupilot-on-osx-with-waf/10618

```
#Note you need to do this from ardupilot code root, not a subdir
./waf configure --board px4-v2 --check-cxx-compiler g++
./waf copter

# if it doesn't build you can clean using
waf distclean
git submodule deinit -f .
```

lvale, 2016-10-08:
Can you please replicate that with a fresh/clean install of MacOS Sierra 10.12 ??

billbonney, 2016-10-08:
The simple answer is no. But i can inform you of some install versions like

```
$ gcc -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/c++/4.2.1
Apple LLVM version 8.0.0 (clang-800.0.38)
Target: x86_64-apple-darwin16.0.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
```
$arm-none-eabi-gcc -v

Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/local/Cellar/gcc-arm-none-eabi/20150925/bin/../lib/gcc/arm-none-eabi/4.9.3/lto-wrapper
Target: arm-none-eabi
Configured with: /Users/build/GCC-4-9-build/gcc-arm-none-eabi-4_9-2015q3-20150921/src/gcc/configure --target=arm-none-eabi --prefix=/Users/build/GCC-4-9-build/gcc-arm-none-eabi-4_9-2015q3-20150921/install-native --libexecdir=/Users/build/GCC-4-9-build/gcc-arm-none-eabi-4_9-2015q3-20150921/install-native/lib --infodir=/Users/build/GCC-4-9-build/gcc-arm-none-eabi-4_9-2015q3-20150921/install-native/share/doc/gcc-arm-none-eabi/info --mandir=/Users/build/GCC-4-9-build/gcc-arm-none-eabi-4_9-2015q3-20150921/install-native/share/doc/gcc-arm-none-eabi/man --htmldir=/Users/build/GCC-4-9-build/gcc-arm-none-eabi-4_9-2015q3-20150921/install-native/share/doc/gcc-arm-none-eabi/html --pdfdir=/Users/build/GCC-4-9-build/gcc-arm-none-eabi-4_9-2015q3-20150921/install-native/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/Users/build/GCC-4-9-build/gcc-arm-none-eabi-4_9-2015q3-20150921/install-native/arm-none-eabi --build=x86_64-apple-darwin10 --host=x86_64-apple-darwin10 --with-gmp=/Users/build/GCC-4-9-build/gcc-arm-none-eabi-4_9-2015q3-20150921/build-native/host-libs/usr --with-mpfr=/Users/build/GCC-4-9-build/gcc-arm-none-eabi-4_9-2015q3-20150921/build-native/host-libs/usr --with-mpc=/Users/build/GCC-4-9-build/gcc-arm-none-eabi-4_9-2015q3-20150921/build-native/host-libs/usr --with-isl=/Users/build/GCC-4-9-build/gcc-arm-none-eabi-4_9-2015q3-20150921/build-native/host-libs/usr --with-cloog=/Users/build/GCC-4-9-build/gcc-arm-none-eabi-4_9-2015q3-20150921/build-native/host-libs/usr --with-libelf=/Users/build/GCC-4-9-build/gcc-arm-none-eabi-4_9-2015q3-20150921/build-native/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-lstdc++ -lm' --with-pkgversion='GNU Tools for ARM Embedded Processors' --with-multilib-list=armv6-m,armv7-m,armv7e-m,cortex-m7,armv7-r
Thread model: single
gcc version 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977] (GNU Tools for ARM Embedded Processors) 
```

```
genromfs -h
genromfs 0.5.2
Usage: genromfs [OPTIONS] -f IMAGE
Create a romfs filesystem image from a directory

  -f IMAGE               Output the image into this file
  -d DIRECTORY           Use this directory as source
  -v                     (Too) verbose operation
  -V VOLUME              Use the specified volume name
  -a ALIGN               Align regular file data to ALIGN bytes
  -A ALIGN,PATTERN       Align all objects matching pattern to at least ALIGN bytes
  -x PATTERN             Exclude all objects matching pattern
  -h                     Show this help

```

lucasdemarchi, 2016-10-15:
@lvale I think we can close this since the improvements would be tracked on #565 and it should be working on mac now just like on Linux. Do you agree?

lvale, 2016-10-15:
done.
And I think that we really must "clean" the dev instructions on all platforms. Lately the problems have been growing.

closed 2016-10-15 by lvale