ArduPilot/ardupilot_wiki #564: Installing APM Planner for Linux - first download link is dead

ncarrier, 2016-10-04:
[Installing APM Planner for Linux](http://ardupilot.org/planner2/docs/installation-for-linux.html) - 
Hello,
Everything is in the title, in the Download section of the "installation for linux" instructions page for APM Planner 2, 2 links are given and the first one sends the user to a page saying "Oops! That page doesn’t exist or is private.".

GregoireH, 2016-10-04:
same for windows

billbonney, 2016-10-04:
Thanks for the comments. it's easy to update the page as well using the edit button, which would also be awesome and very helpful for info you have discovered. 👍 Thanks again

[image]

GregoireH, 2016-10-04:
true, thanks

ncarrier, 2016-10-04:
I'd like to do it, but I'm reading this page because I don't know what is
the correct link 😊

<https://cloud.githubusercontent.com/assets/959994/19081994/2d6961b6-8a11-11e6-88f6-cd1c8b1841a1.png>

GregoireH, 2016-10-04:
is this the corresponding link for windows ?
http://firmware.eu.ardupilot.org/Tools/APMPlanner/apm_planner_2.0.20_win.exe

That's how I got it at least

billbonney, 2016-10-05:
@GregoireH there should be 2.0.24 available now... 

I updated the links to refer to the download locations  for each platform.

I was pointing out the edit button for other issue you may see, and you can help. I realize the download locations for AP2 is really for me to update. Thx 👍

ncarrier, 2016-10-05:
For the Linux part it seems good.
Thank you.

magicrub, 2016-10-05:
Is this ready to be closed?

billbonney, 2016-10-05:
@magicrub Yep, closed

closed 2016-10-05 by billbonney