ArduPilot/ardupilot_wiki #565: Update build instructions

lucasdemarchi, 2016-10-04:
Build instructions are detailed here:
https://github.com/ArduPilot/ardupilot/blob/master/BUILD.md

And here there are some more examples:
examples: https://github.com/ArduPilot/ardupilot/pull/4931#issuecomment-251492778

I think we should have instructions to:
1) Build from command line
2) Build from IDE x, y, z as separate pages

We may add additional pages, previous to these ones with "enviroment setup: Windows, Linux, Mac". I don't think we need to separate per-vehicle. Right now we have conflicting information on different parts of the wiki. What do you think?

hamishwillee, 2016-10-04:
I don't have time to just "take this in hand" so unless others are willing to do all the work, we should do this as a two-phase process.
The first phase would be to add a warning at the top of the [dev build page](http://ardupilot.org/dev/docs/building-the-code.html) that things are changing, and link to the BUILD.md above.

Your idea sounds sane. IFF the environment setup for each platform is relatively straightforward then this could be part of the command line doc. IDE stuff useful but not such a high priority. 

My concern is how to handle all the other docs on the dev build page. Are the instructions still relevant/work in each case, for what versions? If they no longer work we should probably update all of them with a warning up the top pointing to the new build system/docs and stating the last good version that the docs can be used for. 
We should then archive the docs, but provide a link from the new docs stating how you work with old version.

SO, are the old docs relevant, and for what versions?

billbonney, 2016-10-09:
I have added the build.md link to wiki page at the top, so it's more discoverable. I will update the build instructions to macOS(OSX) as well ( soon;) )

https://github.com/ArduPilot/ardupilot_wiki/commit/b6ee75db2d7c227cd4f7ed97275f934d6285417e

magicrub, 2016-10-15:
+1 people are having problems, we need this.

hamishwillee, 2016-10-16:
Yes. Good start. Might be worth going through all the docs on building and adding this note. Though as above, would be even better to add some sort of versioning indicating what version make support ended.

davidbuzz, 2016-11-15:
seems to me that these two pages cover a lot of the same ground:
https://github.com/ArduPilot/ardupilot_wiki/blob/master/dev/source/docs/building-the-code.rst
https://github.com/ArduPilot/ardupilot/blob/master/BUILD.md

hamishwillee, 2016-11-15:
@davidbuzz Not really. The first doc is a grouping topic linking to different platform build instructions using make while the second is specific to building using WAF. Now that WAF is used for building we need to make sure that our docs cover all the required build variations, and then archive all the other make build instructions gracefully so that they can be used for old builds. 

Adding the build.md file is just a basic first step because no one has had time to properly complete the required docs changes.  More discussion on that above: https://github.com/ArduPilot/ardupilot_wiki/issues/565#issuecomment-251530314

khancyr, 2019-01-24:
This should have been fixed ! please reopen if you find that is not the case

closed 2019-01-24 by khancyr