ArduPilot/ardupilot_wiki #569: AutoTune

hpcre, 2016-10-10:
[AutoTune](http://ardupilot.org/copter/docs/autotune.html) - 

On this page the following paragraphs are listed twice 

`AUTOTUNE_AXES allows control of which axis are to be tuned (useful if the vehicle’s battery life is not long enough to complete all 3-axis). “1” = tune roll, “2” = tune pitch, “4” = tune yaw. Add these numbers together to tune multiple axis in a single session (i.e. “7” = tune all axis)

AUTOTUNE_AGGR : Should be in the range of 0.05 to 0.10. Controls the threshold for D-term bounce back and P-term overshoot. This affects the tuning noise immunity (a higher value is more tolerant to flex in the frame or other disturbances that could trick the tuning algorithm). High values also leads to a tune that rejects external disturbances better. Lower values result in a tune that is more responsive to pilot input.
`

magicrub, 2016-10-10:
Thanks @hpcre ! I've edited it in thie PR https://github.com/ArduPilot/ardupilot_wiki/pull/570 and tagged @rmackay9 to take a look before it goes live

davidbuzz, 2016-11-15:
since there's a PR open for this, I think the issue can be closed now as it's adding nothing.

closed 2016-11-15 by davidbuzz