ArduPilot/ardupilot_wiki #571: Build with new mavlink v2.0

futurestark, 2016-10-12:
#include "include/mavlink/v2.0/ardupilotmega/mavlink.h"
I also added  C:\Users\user\Documents\GitHub\ardupilot\modules\PX4Firmware\mavlink\include\mavlink\v2.0\ardupilotmega
and here becouse not sure C:\Users\FutureStark\Documents\GitHub\ardupilot\modules\mavlink\include\mavlink\v2.0
but get error that no such file or directory "include/mavlink/v2.0/ardupilotmega/mavlink.h"

Thank you for your advice

futurestark, 2016-10-12:
And if someone can explaine how debug code with brakepoints hardware on real pixhawk using st-linkv2 or jlink-8, I have installed eclipse on ubuntu and windows can debug stm32f4discovery but don't know how pixhawk

magicrub, 2016-10-12:
This does not appear to be a documentation problem, more of a help question, so it is better to post in discuss.ardupilot.org instead of the wiki issue TODO list.

closed 2016-10-12 by magicrub

gmorph, 2016-10-12:
http://ardupilot.org/dev/docs/jtag.html

futurestark, 2016-10-13:
as i understand from your link it is for APM or ArdupilotMega (AVR boards)  but what about pixhawk where processor is STM32f429 not AVR?