ArduPilot/ardupilot_wiki #573: Building ArduPilot for Pixhawk/PX4 on Linux with Make

atticusliu, 2016-10-17:
Hey guys, I'm having trouble with the following command:
`Tools/scripts/install-prereqs-ubuntu.sh -y`

Which leads to the following error messages:
`E: Unable to locate package gcc-4.6-base:i386
E: Couldn't find any package by glob 'gcc-4.6-base'
E: Couldn't find any package by regex 'gcc-4.6-base'`

I'm running gcc on Ubuntu 5.4.0 (Ubuntu version 16.04.2). I have ran the commands:
`sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
sudo submodule init
sudo submodule update`

Any help is appreciated. I can provide more info as needed. Thanks!

davidbuzz, 2016-11-15:
seems like your system has failed to install gcc for some reason.   I suspect this is not wiki related, and should be brought to the attention of the community at:   http://discuss.ardupilot.org so that 'install-prereqs-ubuntu.sh' can be fixed.

closed 2016-11-15 by davidbuzz

atticusliu, 2016-12-01:
The issue was that I was using Ubuntu 16.04 instead of Ubuntu 14.04. After switching over to 14.04, the issues went away.

magicrub, 2016-12-01:
Hmm.. it should work on 16.x though...