ArduPilot/ardupilot_wiki #578: Outdated port information on SITL

Georacer, 2016-10-19:
Hello everyone!

In [this page](http://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html) on SITL architecture, there is an image of the process communication.

We can see that ArduPlane and JSBSim communicate on ports 5501 and 5502 when if fact the current pair is 5504/5505.

If the picture is meant to be simulator-agnostic, then maybe the constant port numbers should be removed.
This had me scratching my head when I first saw it, due to the inconsistency.

gmorph, 2016-11-15:
I think Tridge would agree with you in removing the port numbers but I prefer to keep them in there as they are indicative and helpful in my opinion.  I have added some information above the picture stating that the port numbers are indicative only and can vary siting your example above.

closed 2016-11-15 by gmorph

Georacer, 2016-11-15:
Cool. Thanks for taking the time to address this.