ArduPilot/ardupilot_wiki #583: Is it possible to convert DataFlash logs to CSV?

doc2142, 2016-10-22:
I want to plot data from the DataFlash log that are in bin format that were pulled from the pixhawk sd card. However, I am having issue finding a way to parse the data in that bin file so that I can plot it in python. Is there a script that will help parse it and convert it to a csv file format?

hamishwillee, 2016-10-23:
This area is for docs issues. Raise this question on the discussion boards please: http://discuss.ardupilot.org/

closed 2016-10-23 by hamishwillee

peterbarker, 2016-10-25:
On Sat, 22 Oct 2016, doc2142 wrote:

> I want to plot data from the DataFlash log that are in bin format that were
> pulled from the pixhawk sd card. However, I am having issue finding a way to
> parse the data in that bin file so that I can plot it in python. Is there a
> script that will help parse it and convert it to a csv file format?

Really the wrong place for this question.  However:
  - there's a "--format csv" option to mavlogdump.py
  - the pymavlink utilities are *written* in Python.  Including, for 
example "mavgraph.py" and "MAVExplorer.py", both useful for plotting 
data..  (https://github.com/ardupilot/pymavlink)

Peter