ArduPilot/ardupilot_wiki #584: Add SidePilot GCS to the common-choosing-a-ground-station page

tommy-b-10, 2016-10-25:
Add the SidePilot GCS to the common-choosing-a-ground-station page in the doc. 
```

SidePilot
---------

ArduCopter and Px4 compatible GCS for iOS Devices.

See support forum for how-to on how connect it to your autopilot

-  **Platform**: iPhone, iPad
-  **Licence**: Proprietary

.. image:: ../../../images/sidepilot.jpg
    :target: ../_images/sidepilot.jpg

-  `Release announcement Blog & Discussion <http://diydrones.com/profiles/blogs/sidepilot-app-version-1-1-formerly-imavlink>`__
-  `Support Forum <http://sidepilot.net/forum>`__
-  `iTunes Store link <https://itunes.apple.com/us/app/sidepilot/id1138193193?ls=1&mt=8>`__
```

Image to show:

[image]

magicrub, 2016-10-25:
Hi @tommy-b-10 it looks like you wrote a pull request inside an issue. Have you tried just passing this code into the wiki directly? If you visit the page you want to edit, at the top right you can "edit in github" and edit then preview right there! If you still need help then we can look at doing that for you

tommy-b-10, 2016-10-25:
Thanks, I was going to do a pull request however I cannot seem to add the image

davidbuzz, 2016-11-15:
Before we add SidePilot to the wiki, and encourage user/s to use it, etc... What sort of feature set does  it have ..?      I don't have the software ( it requires payment ), so I'd appreciate anyone who does have it could go to this document and describe some/all of it's features..?   https://docs.google.com/spreadsheets/d/1PyuViV9NfwAKZbIXa9GBftLn05cTQTtGgdhJ0CjOyXA/edit#gid=2144438540

tommy-b-10, 2016-11-15:
If you would like to test out the software and have a wifi connection to your quad, I'd be happy to give you a promo code for a free version, just chuck me an email. I will update that doc later today when I'm free.

Thanks!

auturgy, 2017-10-15:
reviewing http://ardupilot.org/copter/docs/common-choosing-a-ground-station.html reveals that sidepilot is listed, with the links, image etc.  Closing.

closed 2017-10-15 by auturgy