ArduPilot/ardupilot_wiki #585: Improvement : Document Ardupilot extension to MAVLINK_MSG_ID_MISSION_ITEM

khancyr, 2016-10-27:
Just a reminder to include extension on this message on wiki 
See https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L796

magicrub, 2016-10-27:
none of the commands are covered in the ardupilot wiki, they're in the mavlink docs

closed 2016-10-27 by magicrub

khancyr, 2016-10-27:
@magicrub I put that here because those aren't in mavlink docs, as according to mavlink specification, this parameter should be 0 or 1, and can't has 2 or 3 as value . But ardupilot use them. 
http://mavlink.org/messages/common#MISSION_ITEM
https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L2561

And these command are use in mavproxy and dronekit to make guided action so it seems important to me to reference them

magicrub, 2016-10-27:
fair enough