ArduPilot/ardupilot_wiki #593: LIDAR-Lite Rangefinder RNGFND_SCALING

h3ct0r, 2016-11-05:
[LIDAR-Lite Rangefinder](http://ardupilot.org/copter/docs/common-rangefinder-lidarlite.html) - Hi I have problems with a Lidar Lite v1 using PWM with the RNGFND_SCALING=1. The measured height was approx 30% more than the real one. 

When using 0.8 as scale in RNGFND_SCALING it seems to have better results. I don`t know if the max range to 40m makes a difference when calculating the scale.

hamishwillee, 2016-11-06:
Support queries should be raised on the [discussion boards here](http://discuss.ardupilot.org/). This area is for errors in the documentation.

h3ct0r, 2016-11-06:
@hamishwillee If it wasn't clear: I am showing an error on the documentation. The parameter RNGFND_SCALING=1 as showed in the docs (http://ardupilot.org/copter/docs/common-rangefinder-lidarlite.html) was giving me a 30% error on the measurements. Using RNGFND_SCALING=0.8 seems to fix this problem. 
Maybe this is interesting to put on the docs? I have seen other people complaining about it in forums.

My specific sensor is a LidarLite v1 from PulsedLight3D.

hamishwillee, 2016-11-06:
Thanks @h3ct0r. @rmackay9 Can you please comment on this? Is it just a matter that you should individually calibrate?

magicrub, 2016-11-07:
fixed via https://github.com/ArduPilot/ardupilot_wiki/commit/f7d79240a568f2aa9d67d784af793c0b95ab84ae

closed 2016-11-07 by magicrub