ArduPilot/ardupilot_wiki #599: Intel Aero Overview

patrickpoirier51, 2016-11-12:
[Intel Aero Overview](http://ardupilot.org/rover/docs/common-intel-aero-overview.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

patrickpoirier51, 2016-11-12:
Some users areporting a discripency between  this page and Intel current release product

billbonney, 2016-11-12:
I think you will need to provide the corrected details so they can be applied to the page. Raising an issue against the page won't make the information that is the truth appear... sorry

patrickpoirier51, 2016-11-12:
the intent of  issue is to raise a flag so the information provided is up to date with the actual production board and not related to s preliminary release - if this would be the case after a double check- otherwise  this  issue can be closed

magicrub, 2016-11-12:
Issue is TODO. We don't have any actionable data, do you?

patrickpoirier51, 2016-11-13:
Looking at the getting started  pics 
https://software.intel.com/sites/default/files/managed/25/d5/Intel-Aero-Compute-Board-Getting-Started.pdf
We can clearly see the ms5611 on the back as a confirmation that BOM is consistent with production. Issue closed no action required unless you update wiki with the new pictures

magicrub, 2016-11-13:
Pull requests welcome

patrickpoirier51, 2016-11-14:
Will do once we receive board

closed 2016-11-14 by patrickpoirier51