ArduPilot/ardupilot_wiki #620: Ardupilot.org blog posts can't display videos

rmackay9, 2016-11-30:
When making blog posts on ArduPilot.org the video at the top of the blog posts doesn't show up.

http://ardupilot.org/
http://discuss.ardupilot.org/t/bbc-video-on-outback-challenge-2016/13025

billbonney, 2016-11-30:
The solution for now it to add a image at the top of the post.

like this post http://discuss.ardupilot.org/t/microbes-transported-by-ardupilot-powered-drones-found-safe/11485

OlivierJB, 2017-01-07:
This may be related: External links in a blog post do not  work when clicking directly from the home page of Ardupilot.org (Latest blogs). I suspect there is a bug somewhere in the Concrete5 blogphp processing backend.

rmackay9, 2017-04-25:
I think this is fixed now, thanks Tridge!

closed 2017-04-25 by rmackay9